projectacrn/acrn-hypervisor #2807: Cannot start UOS: ERROR: could not create VCPU 0

Vshmuk, 2019-03-18:
**HW/Board**
NUC model nuc7i7bnh

**Build link**
HV Commit 19c5342506c231cb49a143fc0c1cdbc5cda162a9
acrn-dm version: 0.8-unstable-19c53425 (daily tag:acrn-2019w11.5-140000p),

**Image info**

28310

**Steps**
Using that script:
```
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
		online=`cat $i/online`
		# during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
		while [ "$online" = "1" ]; do
			sleep 1
			echo 0 > $i/online
			online=`cat $i/online`
		done
                echo $idx > ${offline_path}/offline_cpu
        fi
done

/home/clear/opt/acrn-dm -A -m 512M -c 2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  --vsbl /usr/share/acrn/bios/VSBL_debug.bin \
  -s 2,pci-gvt -G "64 448 8" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  --enable_trusty \
  -s 3,virtio-blk,b,/home/clear/launch_android/data/android/android.img \
  -p 0:1 \
  testacrn
```
I see the following:
```
cpu1 online=0
cpu2 online=0
cpu3 online=0
SW_LOAD: get vsbl path /usr/share/acrn/bios/VSBL_debug.bin
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:256/256 page size:0x200000
level 1 free/need pages:1/0 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x20000000, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f9fa7bdd000 -> baseaddr 0x0x7f9fa7c00000
mmap 0x20000000@0x7f9fa7c00000
touch 256 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x20000000, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
polling 33...
             Listening 33...
                            ***********************************************
                                                                           virt-console backend redirected to /dev/pts/4
                                                                                                                        ***********************************************
                                                                                                                                                                       tpm: init_vtpm2:Invalid socket path!
                                                                                                                                                                                                           SW_LOAD: build e820 7 entries to addr: 0x7f9fc7bfd000
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000ef000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000ef000, size 0x0000000000011000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000001ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000020000000, size 0x0000000060000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[5]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000140000000, size 0x0000000000100000,  type 0x2
SW_LOAD: partition blob /usr/share/acrn/bios/VSBL_debug.bin size 3276800 copy to guest 0x3ce0000
                                                                                                SW_LOAD: vsbl_entry 0x3fffff0
                                                                                                                             ERROR: could not create VCPU 0
                                                                                                                                                           mngr_client_new: Failed to accept from fd 33, err: Invalid argument
                                                                                                                                                                                                                              Stop listening 33...
     Stop polling 33...
```

**Expected result**

Start UOS with Android kernel

**Actual result**

Cannot start UOS with Android kernel

**Reproduce rate**
Seems that sometimes it works after reboot, but not so often

Vshmuk, 2019-03-20:
Here is what's inside of 'dmesg':
```
[  335.122674] vhm_dev_open: opening device node
[  335.123077] vhm-ioreq: init request buffer @ 0000000076d954b8!
[  335.123246] vhm-ioreq: created ioreq client 31 for ioeventfd-1
[  335.134178] ACRN vhm ioeventfd init done!
[  335.134332] ACRN vhm irqfd init done!
[  335.134437] vhm: VM 1 created
[  335.189898] region buffer full, set & renew regions!
[  335.190117] region buffer full, set & renew regions!
[  335.197308] vhm-ioreq: created ioreq client 32 for ioreq gvt-g
[  335.200245] acrn-br0: port 1(acrn_tap0) entered blocking state
[  335.200412] acrn-br0: port 1(acrn_tap0) entered forwarding state
[  335.255204] vhm: failed to create vcpu 0!
[  335.277872] vhm-ioreq: destroy ioreq client 32 for ioreq gvt-g
[  335.839812] vhm-ioreq: destroy ioreq client 31 for ioeventfd-1
[  335.839982] vhm-ioreq: client destroying->stop thread
[  335.852339] vhm: freed vm
[  336.203895] acrn-br0: port 1(acrn_tap0) entered disabled state
```

lirui34, 2019-03-21:
How about using default uos script ? 
$ cd /usr/share/acrn/samples/nuc/
$ sudo ./launch_uos.sh

You should change this to your android.img directory previously:
-s 3,virtio-blk,/home/clear/uos/uos.img \
to: 
-s 3,virtio-blk,/home/clear/launch_android/data/android/android.img \

Vshmuk, 2019-03-21:
The same

lirui34, 2019-03-22:
Maybe you can go through this [guide](https://projectacrn.github.io/latest/getting-started/apl-nuc.html#set-up-a-clear-linux-operating-system) or upgrade Clear linux OS to latest and then try UOS again ?

gvancuts, 2019-03-22:
I will see if I can reproduce this on my end, a couple of questions in the meantime:
* What is the SOS kernel version you are using?
I would expect this to happen where there is no physical CPU that's available. The beginning of your output shows that three physical CPUs are *already* offline at the time you start the launch script. CPU0 is used by the SOS.
* Are you running this script multiple times without a reboot?

Note also that the `/dev/acrn_vhm` device node is being renamed to `/dev/acrn_hsm` as part of the upstreaming process (see https://github.com/projectacrn/acrn-hypervisor/commit/13c44f56612dba0dca21dc47a115498c6082322e and https://github.com/projectacrn/acrn-hypervisor/commit/ca3288166646854c9ede2cf7bd4aa4a6c6e59eb7), this is why we have this section at the beginning of the launch script - so that it uses the correct one:
```
offline_path="/sys/class/vhm/acrn_vhm"

# Check the device file of /dev/acrn_hsm to determine the offline_path
if [ -e "/dev/acrn_hsm" ]; then
offline_path="/sys/class/acrn/acrn_hsm"
fi
```

Can you check in your set-up what device node is used and make sure that your launch script (esp. the `offline_path` variable) uses the correct one?

gvancuts, 2019-03-22:
One more thought: the way the script runs, it will only perform the `echo $idx > ${offline_path}/offline_cpu` step *if* the CPU was online initially (which is not the case for you for some reason). This step is required for ACRN to be able to start a VM. Can you clarify why the CPUs are all already offline and if that step was performed at any point in time prior to launching the script?

Vshmuk, 2019-03-23:
>     * What is the SOS kernel version you are using?
Linux clr-sos-guest 4.19.28-28.iot-lts2018-sos #1 SMP PREEMPT Wed Mar 13 10:48:19 UTC 2019 x86_64 GNU/Linux
>       I would expect this to happen where there is no physical CPU that's available. The beginning of your output shows that three physical CPUs are _already_ offline at the time you start the launch script. CPU0 is used by the SOS.

That's obviously because i've already run the script after system reboot. The CPUs are already offline.

>     * Are you running this script multiple times without a reboot?
This.

> Note also that the `/dev/acrn_vhm` device node is being renamed to `/dev/acrn_hsm` as part of the upstreaming process (see [13c44f5](https://github.com/projectacrn/acrn-hypervisor/commit/13c44f56612dba0dca21dc47a115498c6082322e) and [ca32881](https://github.com/projectacrn/acrn-hypervisor/commit/ca3288166646854c9ede2cf7bd4aa4a6c6e59eb7)), this is why we have this section at the beginning of the launch script - so that it uses the correct one:
> 
> ```
> offline_path="/sys/class/vhm/acrn_vhm"
> 
> # Check the device file of /dev/acrn_hsm to determine the offline_path
> if [ -e "/dev/acrn_hsm" ]; then
> offline_path="/sys/class/acrn/acrn_hsm"
> fi
> ```
> 
> Can you check in your set-up what device node is used and make sure that your launch script (esp. the `offline_path` variable) uses the correct one?

```
$ ls -la /dev/acrn_
acrn_hvlog_cur_0  acrn_hvlog_cur_3  acrn_trace_2      
acrn_hvlog_cur_1  acrn_trace_0      acrn_trace_3      
acrn_hvlog_cur_2  acrn_trace_1      acrn_vhm     
$ ls /sys/class/vhm/acrn_vhm/
dev  offline_cpu  power  subsystem  uevent
$ ls -la /sys/class/acrm
ls: cannot access '/sys/class/acrm': No such file or directory
```

Vshmuk, 2019-03-23:
Well, maybe i've localized a bug.
I have two scripts. The first one is this:
```
#!/bin/bash
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
		online=`cat $i/online`
		# during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
		while [ "$online" = "1" ]; do
			echo "CPU $i setting to offline"
			sleep 1
			echo 0 > $i/online
			online=`cat $i/online`
		done
                echo $idx > ${offline_path}/offline_cpu
        fi
done

/home/clear/opt/acrn-dm -A -m 512M -c 2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  --vsbl /usr/share/acrn/bios/VSBL_debug.bin \
  -s 2,pci-gvt -G "64 448 8" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  --enable_trusty \
  -s 3,virtio-blk,b,/home/clear/launch_android/data/android/android.img \
  -p 0:1 \
  testacrn
```
The second one:
```
#!/bin/bash

offline_path="/sys/class/vhm/acrn_vhm"

# Check the device file of /dev/acrn_hsm to determine the offline_path
if [ -e "/dev/acrn_hsm" ]; then
offline_path="/sys/class/acrn/acrn_hsm"
fi

function launch_clear()
{
mac=$(cat /sys/class/net/e*/address)
vm_name=vm$1
mac_seed=${mac:9:8}-${vm_name}

#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

#for memsize setting
mem_size=512M

/home/clear/opt/acrn-dm -A -m $mem_size -c $2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 2,pci-gvt -G "$3" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  -s 3,virtio-blk,/home/clear/uos/uos.img \
  -s 4,virtio-net,tap0 \
  --mac_seed $mac_seed \
  -k /usr/lib/kernel/default-iot-lts2018 \
  -B "root=/dev/vda3 rw rootwait maxcpus=$2 nohpet console=tty0 console=hvc0 \
  console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M \
  consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=$4 \
  i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0 \
  i915.enable_guc_submission=0 i915.enable_guc=0" $vm_name
}

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
		# during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
		while [ "$online" = "1" ]; do
			sleep 1
			echo 0 > $i/online
			online=`cat $i/online`
		done
                echo $idx > ${offline_path}/offline_cpu
        fi
done

launch_clear 1 1 "64 448 8" 0x070F00
```
The most noticable difference between these two scripts is that the CPU offline loop executes immediately in the first script, and it takes 4 seconds to complete the same code in the second script (where the code waits longer and runs "sleep 1" 4 times). This looks very odd for me, since these two loops are fully identical. I can admit that there may be a typo or a mistake in my code, but the problem is that the order of running is important. Something in the first script influences the system so that the second script cannot be run. But if i run the second script before first one, both of them start working!

Vshmuk, 2019-03-23:
Update: yes, i have a typo in the first script, where $offline_path is not defined. After that, the line "$offline_path="/sys/class/acrn/acrn_hsm" is never executed even in the correct script. So, we can consider it's not a bug, but maybe it's okay to add some additional checks in the reference script to avoid such situation?

Mingyuan18, 2019-03-28:
[External_System_ID] ACRN-3203

gvancuts, 2019-03-28:
> Update: yes, i have a mistake in the first script, where $offline_path is not defined. After that, the line `echo $idx > /sys/class/acrn/acrn_hsm/offline_cpu` is never executed even in the correct script. So, we can consider it's not a bug, but maybe it's okay to add some additional checks in the reference script to avoid such situation?

The scripts are a bit rough around the edges in general, I'm not the decider but I would welcome additional checks or warnings when something unusual is detected.

It's also been bugging me a bit that we unconditionally take **all** CPUs offline (except the BSP), irrespective of what we want to give to the VMs (and by default we only give one CPU to a VM). The scripts are also not well suited to started multiple VMs and need to manually edited before doing so.

wenlingz, 2019-04-04:
@Vshmuk @gvancuts This is not a bug, but need optimize, so could we can close this issue now ?thanks.

gvancuts, 2019-04-04:
I'm OK closing this. If @Vshmuk wanted to send some PR to improve the scripts, we could re-open this and use it in the `Tracked-On:` field.

Mingyuan18, 2019-04-08:
OK, so close this issue first.

closed 2019-04-08 by Mingyuan18

Vshmuk, 2019-04-12:
Hi! Sorry, i had a vacation. Do you still need any PRs from me?