projectacrn/acrn-hypervisor #2830: Warning when building the hypervisor

gvancuts, 2019-03-20:
Since commit 334382f9ef1f0564598cd2f96ad22d1a7f8b4c1c , compiling the ACRN hypervisor for EFI (`make BOARD=apl-nuc FIRMWARE=uefi`) throws some (non-fatal) warnings. While harmless, we should fix these.

Extract of the output that shows the warnings:
```
gcc  -I/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor/include -I. -c -I. -I.. -I../hypervisor/include/arch/x86/guest -I/usr/include/efi -I/usr/include/efi/x86_64 -I../hypervisor/include/public -I../hypervisor/include/lib -I../hypervisor/bsp/include/ -DEFI_FUNCTION_WRAPPER -fPIC -fshort-wchar -ffreestanding -Wall -I../fs/ -Dx86_64 -O2 -I../hypervisor/include/arch/x86 -include config.h -mno-mmx -mno-sse -mno-sse2 -mno-80387 -mno-fp-ret-in-387 -mno-red-zone  malloc.c -o /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/malloc.o
In file included from ../hypervisor/bsp/include/firmware_uefi.h:10,
                 from boot.c:40:
../hypervisor/bsp/include/firmware.h:16:38: warning: ‘struct acrn_vm’ declared inside parameter list will not be visible outside of this definition or declaration
  int32_t (*init_vm_boot_info)(struct acrn_vm *vm);
                                      ^~~~~~~
../hypervisor/bsp/include/firmware.h:24:43: warning: ‘struct acrn_vm’ declared inside parameter list will not be visible outside of this definition or declaration
 int32_t firmware_init_vm_boot_info(struct acrn_vm *vm);
                                           ^~~~~~~
In file included from boot.c:40:
../hypervisor/bsp/include/firmware_uefi.h:22:48: warning: ‘struct acrn_vm’ declared inside parameter list will not be visible outside of this definition or declaration
 int32_t uefi_init_vm_boot_info(__unused struct acrn_vm *vm);
                                                ^~~~~~~
ld -T /usr/lib64/elf_x86_64_efi.lds -Bsymbolic -shared -nostdlib -znocombreloc -L/usr/lib64 /usr/lib64/crt0-efi-x86_64.o -o /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/boot.so /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/boot.o /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/pe.o /home/gvancuts/work/acrn/acrn-hypervisor/build/efi-stub/malloc.o  -lgnuefi -lefi /usr/lib64/gcc/x86_64-generic-linux/8.3.1/libgcc.a
```

Mingyuan18, 2019-03-21:
[External_System_ID] ACRN-3073

gvancuts, 2019-03-21:
#2832 solves the issue and was merged, closing this issue.

Mingyuan18, 2019-03-22:
OK,close it.

closed 2019-03-22 by Mingyuan18