projectacrn/acrn-hypervisor #2848: Cannot boot SOS kernel outside of ACRN

rossburton, 2019-03-25:
If I boot the SOS kernel outside of ACRN, it panics on startup.  The panic log looks like this:

```
? do_early_param
? early_hvlog
acrn_hvlog_init
? acrn_trace_init
? early_hvlog
? do_early_param
do_one_initcall
..
kernel_init
```

I can provide a photo of the screen, but the NUC I'm replicating this on doesn't have serial and it crashes before netconsole is up.

Mingyuan18, 2019-03-26:
@rossburton Can you provide more detailed reproduce steps?

gvancuts, 2019-03-26:
I think it's as simple as:
1. Install Clear Linux (native) on a NUC
2. Add the `service-os` bundle
2. Reboot and select the `sos` kernel
(I can double-check that later but I'm pretty sure that's it)

rossburton, 2019-03-26:
Pretty much.  You might need to pass the hvlog parameter to the kernel.  Note that this used to work, for a while now our startup procedure is:

1. Boot SOS kernel outside of ACRN
2. First boot script edits EFI configuration, adding a ACRN entry
3. Reboot, ACRN starts and boots SOS kernel

Since a recent update (1) is now failing with a kernel panic.

Mingyuan18, 2019-03-28:
[External_System_ID] ACRN-3196

fyin1, 2019-04-12:
We found the root cause of this issue. And working on fix.

Mingyuan18, 2019-04-16:
@rossburton , this PR's acrn tag is : acrn-2019w16.1-140000p ; please help to confirm whether this issue pass or not. Thanks!

fyin1, 2019-04-16:
@rossburton Please note, you only need to update the kernel part because the fixing is in kernel. Thanks.

Mingyuan18, 2019-04-23:
@rossburton , We have verified this issue and confirmed it was fixed. If you haven't any reply after a week, we will close this issue. Thank yo

rossburton, 2019-05-07:
Sorry was away.  I can confirm this is fixed.

closed 2019-05-07 by rossburton