projectacrn/acrn-hypervisor #2849: hv: hypervisor will hang in fuzzing HV_VM_SET_MEMORY_REGIONS hypercall

yonghuah, 2019-03-26:
In hypervisor fuzzing test, hypervisor will hang  if issuing 
HV_VM_SET_MEMORY_REGIONS hypercall after  target VM is destroyed.

We used Syzkaller ran with hypercall unit tests to do Fuzzing test for ACRN, which ran on SOS and communicate with DM process by socket.
enable_syscalls: ["syz_ic_get_api_version", "syz_ic_create_vm", "syz_ic_destroy_vm", "syz_ic_start_vm", "syz_ic_pause_vm", "syz_ic_reset_vm", "syz_ic_create_vcpu" ]
1. Reproduce Steps
--------------------------------------------------------------
setup env with wiki: And sync latest ACRN code and fuzzing tool code to your host
apply patch_for_fuzzing_on_dm.txt to devicemodel, and build images
modify acrn_build.sh based your own environment, and run it to rebuild syzkaller tool
flash images, and then make uos autoboot, remove sos password, crashlogctl enable
from fuzzing tool repo, cp syzkaller/bin/linux_amd64/syz-execprog and syz-executor into SOS
in SOS, reproduce with crash log(as attchment acrn_general-vmm_running3.log) "./syz-execprog -executor=./syz-executor -repeat=0 -procs=16 -cover=0 **run.log"
2. Expected result:
--------------------------------------------------------------
SOS/UOS not crashed not hang, and SUT works well
3. Current result:
--------------------------------------------------------------
in SOS, reproduce with crash log "./syz-execprog -executor=./syz-executor -repeat=0 -procs=16 -cover=0 **run.log"
DUT HV crash
Hypervisor crash logs as below, detail as attachment log(HVcrash_pagetable.log)
root@clr-9c5fc17c8c9a4ff486962ab359f54fab ~/tmp # ./syz-execprog -executor=./syz-executor -repeat=0 -procs=16 -cover=0 acrn_general-vmm_running3.log
2019/02/25 04:23:17 parsed 8 programs
2019/02/25 04:23:17 executed programs: 0
[14526363489us][cpu=1][sev=1][seq=75]:emulate_io Err: access dir 1, type 0, addr = 0x400, size=2
[14526373123us][cpu=1][sev=1][seq=76]:dispatch VM exit handler failed for reason 30, ret = -22!
2019/02/25 04:23:22 executed programs: 96
2019/02/25 04:23:28 executed programs: 192
2019/02/25 04:23:34 executed programs: 288
2019/02/25 04:23:39 executed programs: 384
2019/02/25 04:23:44 executed programs: 472
2019/02/25 04:23:49 executed programs: 560
Assertion failed in file arch/x86/pagetable.c,line 226 : fatal error
Host Stack: CPU_ID = 0
addr(0x74c82d88) 0x0000000074ca0000 0xffffffffffffffff 0x0000000073836000 0x0000000081800000
addr(0x74c82da8) 0x0000000074c82e48 0x000000006e004125 0x0000000000000000 0x0000000073836000
addr(0x74c82dc8) 0x0000008000000000 0xffffffffffffffff 0x0000000000000000 0x0000000074c82e4c
addr(0x74c82de8) 0x0000000074c82e28 0x8ee2bc38f5188720 0x0000000040000000 0x8ee2bc38f5188720
5. repetition rate
----------------------------------------------------------------
5/5

Mingyuan18, 2019-03-26:
[External_System_ID] ACRN-2862

Mingyuan18, 2019-04-08:
This issue was fixed now:
ACRN tag:acrn-2019w12.3-150000p
SOS: 28100
Devices: MRB

closed 2019-04-08 by Mingyuan18