projectacrn/acrn-hypervisor #2851: [APL/KBL/UP2][HV][LaaG]Uos cannot boot when acrnctl add Long_VMName of more than 26

taoyuhong, 2019-03-26:
1.Environment
[Board]: APL NUC&KBL NUC&APL UP2

Installed version: 28100
Version URL: https://cdn.download.clearlinux.org/update/
Content URL: https://cdn.download.clearlinux.org/update/
root@sos-dom0 ~ # uname -a
Linux sos-dom0 4.19.23-23.iot-lts2018-sos #1 SMP PREEMPT Thu Feb 28 07:19:44 UTC 2019 x86_64 GNU/Linux
root@sos-dom0 ~ # acrn-dm -v
DM version is: 0.7-unstable-▒”2019w09.5.140000p_167▒” (daily tag:▒”2019w09.5.140000p▒”), build by mockbuild@2019-03-01 13:22:58

2.Reproduce step:
--------------------------------------------------------------
1). Boot to SOS.
2). Modify vm_name in launch_uos.sh to vm123456789012345678901234567
3). add vm by
acrnctl add launch_uos.sh
4). list vm by 
acrnctl list
5). Start vm by
acrnctl start vm123456789012345678901234567

3. Current result:
--------------------------------------------------------------
Uos cannot boot when acrnctl add Long_VMName of more than 26.

4. Expected result:
--------------------------------------------------------------
Uos boot successfully when acrnctl add Long_VMName of more than 26.

5. repetition rate
--------------------------------------------------------------
100%

6.Logs:

root@sos-dom0 ~ # acrnctl start vm123456789012345678901234567
ERROR: command is truncated

[External_System_ID] ACRN-2921

acrnsi, 2019-03-26:
[External_System_ID] ACRN-2921

Mingyuan18, 2019-04-04:
Can't reproduce this issue in below build:
Installed version: 28600
ACRN tag:”2019w14.1.140000p”

closed 2019-04-04 by Mingyuan18