projectacrn/acrn-hypervisor #2854: UOS occasionally failed to boot

kiendinh, 2019-03-27:
**HW/Board**
NUC model nuc7i7dnhe

**ACRN version**
Acre release v0.4

**Clear Linux info**
26760

**Launch script**
`# for memsize setting
    mem_size=4096M
    acrn-dm -A \
            -m $mem_size \
            -c $2 \
            -s 0:0,hostbridge \
            -s 1:0,lpc -l com1,stdio \
            -s 2,pci-gvt -G "$3" \
            -s 5,virtio-console,@pty:pty_port \
            -s 6,virtio-hyper_dmabuf \
            -s 3,virtio-blk,/home/clear/agl-rse.wic \
            -s 4,virtio-net,$tap \
            -s 7,xhci,1-5 \
            -k /home/clear/bzImage \
            -B "root=/dev/vda2 rw rootwait maxcpus=$2 nohpet console=tty0 console=hvc0 \
              console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M \
              consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=$4 \
              i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0 \
              i915.enable_guc_submission=0 i915.enable_guc=0" $vm_name
}

# offline SOS CPUs except BSP before launch ivi UOS
for i in `ls -d /sys/devices/system/cpu/cpu[2-4]`; do
    online=`cat $i/online`
    idx=`echo $i | tr -cd "[2-4]"`
    echo cpu$idx online=$online
    if [ "$online" = "1" ]; then
        echo 0 > $i/online
        echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
fi done

launch_agl 2 1 "64 448 8" 0x070000 "cluster"`

**Expected result**
VM launches

**Actual result**
VM sometimes failed to launch. The repetition rate is low. Once or twice a day in a regular basic development.

**Log**
clear@clr ~ $ 
clear@clr ~ $ sudo ./launch_rse.sh 
cpu2 online=0
cpu3 online=0
cpu4 online=0
tap device existed, reuse acrn_tap1
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get kernel path /home/clear/bzImage
SW_LOAD: get bootargs root=/dev/vda2 rw rootwait maxcpus=1 nohpet console=tty0 console=hvc0 			console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M 			consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=0x070000 			i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0 			i915.enable_guc_submission=0 i915.enable_guc=0
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643

open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643

level 0 free/need pages:791/0 page size:0x200000
level 1 free/need pages:0/4 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 10): echo 14 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
to reserve pages (+orig 1815): echo 3072 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
level 0 pages gap: 4 failed to reserve!
Unable to setup memory (0)
clear@clr ~ $

Mingyuan18, 2019-03-27:
@kiendinh Can you try this issue with the latest ACRN release V0.7? Thanks!

kiendinh, 2019-04-01:
I've tried with ACRN v0.7 and it didn't even boot anymore.

===
cpu2 online=0
cpu3 online=0
cpu4 online=0
cpu5 online=0
cpu6 online=0
cpu7 online=0
acrn-br0 bridge aleady exists, adding new tap device to it...
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get kernel path /root/bzImage-4.19.8-uos-20190125
SW_LOAD: get bootargs root=/dev/vda2 rw rootwait maxcpus=1 nohpet console=tty0 console=hvc0     console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M     consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=0x070000     i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0     i915.enable_guc_submission=0 i915.enable_guc=0
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/0 page size:0x200000
level 1 free/need pages:4/2 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7fc20861e000 -> baseaddr 0x0x7fc240000000
mmap 0x80000000@0x7fc240000000
touch 2 pages with pagesz 0x40000000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
Listening 0...
polling 0...
***********************************************
                                               virt-console backend redirected to /dev/pts/2
                                                                                            ***********************************************
            tpm: init_vtpm2:Invalid socket path!
                                                /tmp/dm.XKQC8Q4     30: [0004]		Address : fec00000
                                                                                                          Error    6302 -  Flag value is too large ^  (Maximum 1 bit)

                                      /tmp/dm.XKQC8Q4     37: [0004]		Interrupt : 00000002
                                                                                                    Error    6302 -    Flag value is too large ^  (Maximum 1 bit)

                                  /tmp/dm.XKQC8Q4     46: [0004]		Interrupt : 00000009
                                                                                                    Error    6302 -    Flag value is too large ^  (Maximum 2 bit)

                                  /tmp/dm.XKQC8Q4     47: [0002]		Flags (decoded below) : 000D
                                                                                                            Error    6302 -                Flag value is too large ^  (Maximum 2 bit)

                                                      vtnet tx thread closing...
                                                                                mngr_client_new 113
                                                                                                   Stop listening 33...
                                                                                                                       Stop polling 33...

gvancuts, 2019-04-01:
This is very similar than https://github.com/projectacrn/acrn-hypervisor/issues/2795. You need to make sure that the `iasl` tool version matches your ACRN.

kiendinh, 2019-04-01:
Thank you @gvancuts! I upgraded the system to CL v28560 with iasl 20190215. So far the VM could boot and reboot. It's the rarely occurring bug so we'll keep testing for a while. By the way, I hope that they will add some system checks in ACRN to resolve the dependency on iasl.

Mingyuan18, 2019-04-02:
@kiendinh,  If you can't reproduce this problem, could you close this bug ? Thanks!

kiendinh, 2019-04-02:
It is a bug that happens occasionally, and it may take longer period to see if it is still there. Also, it's being tested in the customer environment, and thus I would like to keep it until they verify that everything is fine.

Mingyuan18, 2019-04-02:
@kiendinh Ok, let's keep paying attention to it, thank you！

kiendinh, 2019-04-02:
By the way @Mingyuan18, I'm curious if there is another way to solve bug instead of getting to the latest release. As you know the customer tend to stay with a stable release when most of the features required work. There are probably risks upgrading to a newer version without going through the tests again. And that's not usually easy. What is the dev model of ACRN for development support?

hwang37, 2019-04-02:
Thanks for your question. I totally understand the requirement difference between upstream project and customer product. One is seeking for latest feature while the other direction is to stabilize existing the features.

1. For ACRN, our Master branch is working as development branch, always merging newest features. We'll guarantee the new feature patches not break other existing features, but it takes time to become mature. 
2.  Regularly we branch out a branch named "ver x.y" for release purpose, having more validations on it. But it's for release reference not long term sustaining commitment. 
3. Community developer can select one of release branch as their preferred development baseline, back port new features or bug fixes from Master branch.
4. Any release branch bug reported by community developers, we'll verify it on Master branch first, if it's reproducible in Master, we'll fix in Master.

kiendinh, 2019-04-04:
Thank you so much for the explanations, @hwang37.

kiendinh, 2019-04-08:
Unfortunately we're still having the same issue after upgrading to ACRN v0.7 & ClearLinux v28560. Though it is only occasional, following the [guide](https://projectacrn.github.io/latest/tutorials/agl-vms.html) on ACRN v0.7 & Clear Linux 28560, the phenomenon can be reproduced just by repeating these steps.

- Launch the IVI VM by using launch_ivi.sh
- Launch the RSE VM by using launch_rse.sh
- shutdown IVI VM
- Relaunch the IVI VM again using launch_ivi.sh

gvancuts, 2019-04-08:
> * shutdown IVI VM

Out of curiosity, how are you shutting down the IVI VM?

@hwang37 , I'm like @kiendinh and always use the scripts directly (and shut down the VMs by using `poweroff` from inside the VM). What is the engineering's team recommendation with regards to using the [acrnctl](https://projectacrn.github.io/latest/tools/acrn-manager/README.html#acrnctl-and-acrnd) tool instead, does it do anything differently?

kiendinh, 2019-04-08:
@gvancuts We just simply used "poweroff" from inside the VM.

Mingyuan18, 2019-04-09:
[External_System_ID] ACRN-3334

hwang37, 2019-04-09:
@JasonChenCJ @fyin1  to comment Geoffroy's question about acrnctl to control VM shutdown

hwang37, 2019-04-10:
Is there memory leak here, in terms of "don't release hugepage"?

fyin1, 2019-04-10:
I believe the hugepage for previous UOS instance was not free cleanly. This could be function issue instead of memory leak.

libing11, 2019-04-19:
@kiendinh :

1： when the problem occurs, run the launch_ivi.sh, can you continue to start the virtual machine?
2：Did you start two virtual machines at the same time?
3:  Are the contents of these two scripts the same? (launch_ivi.sh, launch_rse.sh)
4:  Can you leave your email?

kiendinh, 2019-04-19:
@libing11 
1. No, once it failed, it could not be relaunched. We needed to reboot the machine and then launch the VMs again.
2. We launched IVI VM using launch_ivi.sh. After it started successfully, we then launched the RSE using launch_rse.sh.
3. The launch scripts are similar but not the same. You can take a look at them here:
https://projectacrn.github.io/latest/tutorials/agl-vms.html
4. I'll contact you via internal channel.

libing11, 2019-05-31:
This issue has been identified, the specific reasons are:
If you run two acrn-dm processes at the same time, hugetblfs will be mounted twice, which will cause a memory leak.
I have already made a patch, after testing, the problem has disappeared.

Mingyuan18, 2019-07-19:
@kiendinh , Please confirm if this issue has been resolved. The patch was fixed. Thanks!

fuzhongl, 2019-07-22:
@kiendinh 
Patch merged in CL 30210.
Please help to verify this issue, if no feedback within two weeks, the ticket will be closed by default.
Thanks for your understanding!

closed 2019-08-05 by Mingyuan18