projectacrn/acrn-hypervisor #2857: FAQs for ACRN's memory usage need to be updated

kriskep, 2019-03-27:
The FAQs that describe the configuration changes needed for a 32G system do not work on v0.7 anymore.
ref. https://projectacrn.github.io/latest/faq.html#how-do-i-configure-acrn-s-memory-use

Applying the settings from above link gives a build error.

Build Error:
```
/usr/bin/ld: /home/kris/acrn-hypervisor/build/hypervisor/acrn.out section `.bss' will not fit in region `ram'
/usr/bin/ld: region `ram' overflowed by 369648 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:342: /home/kris/acrn-hypervisor/build/hypervisor/acrn.out] Error 1
make[1]: Leaving directory '/home/kris/acrn-hypervisor/hypervisor'
make: *** [Makefile:50: hypervisor] Error 2
```

Increasing the HV_RAM_SIZE to 0x0f100000 gets rid of the build error, but the system still hangs at the UEFI boot.

Mingyuan18, 2019-03-28:
[External_System_ID] ACRN-3181

wenlingz, 2019-04-04:
@kriskep sorry for the inconvenient, we will create the PR to modify the FAQ for the ACRN memory usage.
If the SOS enlarge to the 32G, UOS RAM enlarge to the 16G, you  need modify it with below formula:
(delta +512G -1bit)/512G +(delta+1G-1bit)/1G +(dalta+2M-1bit)/2M pages, and the delta is (SOS+UOS)=24G, SOS:(32G-16G), UOS:(16G-8G), so delta=24G, where pages are 4K. Besides HV_RAM_SIZE must be 2M align.
@lifeix please help review above, and comments, thanks.

wenlingz, 2019-04-10:
@kriskep please refer our comment, if works, maybe please help close this issue, we will modify the related FAQ

Mingyuan18, 2019-04-11:
@kriskep please refer our comment, if works, maybe please help close this issue, we will modify the related FAQ

Mingyuan18, 2019-04-16:
@kriskep ,Do you have any update? Thanks!

kriskep, 2019-04-16:
that is a rather complicated formula. Not sure my calculation is right, but I get:
1+24+12288 = 12313 (pages of 4k), so that is 0x3019000.
Added that to the default of 0xb800000  gives 0xe819000.
rounded to align to 2M: 0xea00000
so I tried with HV_RAM_SIZE=0x0ea00000, but that fails to build the hypervisor:

/usr/bin/ld: /home/kris/acrn-hypervisor/build/hypervisor/acrn.out section `.bss' will not fit in region `ram'
/usr/bin/ld: region `ram' overflowed by 6597624 bytes

kriskep, 2019-04-16:
HV_RAM_SIZE=0x0f200000 seems to work.

not sure what the formula is. I just added the missing 6597624 bytes to what I already had, then rounded up to align to 2M.
It would be nice if the HV_RAM_SIZE is calculated automatically at build time.

wenlingz, 2019-04-17:
> HV_RAM_SIZE=0x0f200000 seems to work.
> 
> not sure what the formula is. I just added the missing 6597624 bytes to what I already had, then rounded up to align to 2M.
> It would be nice if the HV_RAM_SIZE is calculated automatically at build time.

wenlingz, 2019-04-17:
@lifeix please help give kriskep details explanation about the formula, thanks a lot

lifeix, 2019-04-17:
@kriskep 
Hi Kriskep. 

Now we use static memory to manage guest EPT mapping for FuSa. So we need to reserve memory in ACRN hypervisor for EPT mapping. While EPT has 4 level page table. Level-1 could hold 512 x 512G memory; Level -2 could hold 512 x 1G memory; Level -3 could hold 512 x 2M; Level -4 could hold 512 x 4K. 
So if the board memory has enlarged, we need to reserve more memory for EPT mapping. If you enlarge the board memory from 16G to 32G, you should reserve one more page for Level-1 EPT page table; one more page for Level-2 page table;  16 more pages for Level-3 page table and 8192 more pages for Level-4 page table.  This is just only for SOS. For **a** UOS, if you don't plan to enlarge its memory. It's done. If you want to enlarge its memory too, you need to reserve more memory for it to manage its EPT mapping too. 
Besides all of this,  we doesn't use Level-4 page table in ACRN hypervisor. The minimal granularity  of page size is 2 MB. So the HV_RAM_SIZE  should aligned to 2 MB too. 

1) not sure what the formula is. I just added the missing 6597624 bytes to what I already had, then rounded up to align to 2M.
I think you configure more than one UOS in your arch/x86/config
2) It would be nice if the HV_RAM_SIZE is calculated automatically at build time.
It will be used in bsp/ld/link_ram.ld.in. It needs to be an constant. If you doesn't pay attention on memory used by hypervisor, just configure it large enough to manage EPT mapping seems something easier to do, like 1/16 of the boards memory (Of course, 2 MB aligned).

fuzhongl, 2019-05-24:
@kriskep Any more question? Could you please close this ticket?
Thanks!

kriskep, 2019-05-24:
Thanks for the clarification.

closed 2019-05-24 by kriskep