projectacrn/acrn-hypervisor #2886: It will reminder:"/run/acrn/mngr: No such file or directory" when run acrnctl command.

Mingyuan18, 2019-04-04:
1.Environment
[Board]: APL NUC&KBL NUC&APL UP2
Installed version: 28340
root@sos-dom0 ~ # uname -a
Linux sos-dom0 4.19.28-28.iot-lts2018-sos #1 SMP PREEMPT Wed Mar 13 10:48:19 UTC 2019 x86_64 GNU/Linux
root@sos-dom0 ~ # acrn-dm -v
DM version is: 0.8-unstable-”2019w11.5.140000p_173” (daily tag:”2019w11.5.140000p”), build by mockbuild@2019-03-15 14:26:40

2.Reproduce step:
1). Boot to SOS.
2). Run command:acrnctl list

3. Current result:
It will reminder:"/run/acrn/mngr: No such file or directory" when run acrnctl list command.

4. Expected result:
Expect without any fail and error when run the command.

5. repetition rate
100%

6.Logs:
root@sos-dom0 ~ # acrnctl list
/run/acrn/mngr: No such file or directory
_scan_alive_vm: Failed to check directory /run/acrn/mngr, err: -1
vm1             stopped

Mingyuan18, 2019-04-04:
[External_System_ID] ACRN-3055

Mingyuan18, 2019-04-29:
This is not a bug. Current Basic Framework Design for Implementation. Before execute the command: acrnctl list, you should confirm that the service of acrnd had started in backgroud. You can check it use command: ps -ef |grep acrnd.  If acrnd is not started, you should start service like this: acrnd &.

closed 2019-04-29 by Mingyuan18

gvancuts, 2019-04-29:
A nice improvement would be to display a message that gives the user a hint to check that, something like: `Please make sure that acrnd is running`.