projectacrn/acrn-hypervisor #2891: "Using Ubuntu as the Service OS" not working on my NUC7i7BNH & NUC6i7KYK

RoboticsYY, 2019-04-04:
Hi, I am attempting to get the ACRN Hypervisor to function on my **NUC7i7BNH** & **NUC6i7KYK**. But the efforts on both of them failed.

I have followed the steps in [Using Ubuntu as the Service OS](https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html) to set up Ubuntu as Service OS.

I have updated the BIOS to the latest version. I have successfully set up Ubuntu 18.04 and build ACRN from source, and followed the rest steps to install the hypervisor and service os kernel. Then reboot. 

* For my **NUC7i7BNH**: the booting of Ubuntu stuck after the grub screen.
* For my **NUC6i7KYK**: the system stuck at screen "detected 8 processes, 8 enabled current on process 0"

I have repeated the process on the two NUCs several times, I'm quite sure I have made no mistakes to the commands. Could anyone help me on that? I'm based on PRC zizhu. Since my project timeline is very urgent, If there is someone from ACRN team also based on this site, can I connect to someone for direct help? Thanks.

gvancuts, 2019-04-04:
Hi @RoboticsYY , can you confirm how much memory you have on each system? (we have seen issues booting on systems with 32GB of RAM, we typically test more with 8 or 16GB).

What version of ACRN and the Service OS kernel are you using?

The **NUC6i7KYK** is not officially tested so I'd recommend we focus on the **NUC7i7BNH** device for now.

RoboticsYY, 2019-04-06:
Hi @gvancuts 

> Hi @RoboticsYY , can you confirm how much memory you have on each system? (we have seen issues booting on systems with 32GB of RAM, we typically test more with 8 or 16GB).

The RAM of **NUC7i7BNH**  is 8G (4Gx2)

> What version of ACRN and the Service OS kernel are you using?

Previously, I used `org.clearlinux.iot-lts2018-sos.4.19.0-22` as the tutorial hints. I just made a retry with `org.clearlinux.iot-lts2018-sos.4.19.32-42`. After the grub screen, the boot did not get stuck, but it also failed with the following information popped up:

[image]

lirui34, 2019-04-08:
Hi @RoboticsYY ,  could you please double check this file: /etc/grub.d/40_custom if you're setting the correct root partition or not ?

RoboticsYY, 2019-04-08:
Hi @lirui34, I just checked /etc/grub.d/40_custom, I think I have made a mistake by using `PARTUUID` of boot partition not the root partition. Now, I can boot into ACRN Hypervisor and see Ubuntu desktop. Thanks for your Help!

gvancuts, 2019-04-08:
Thanks for the update @RoboticsYY , I'm closing this issue, feel free to open a new one if you encounter further problems.

closed 2019-04-08 by gvancuts