projectacrn/acrn-hypervisor #2901: Plus and minus signs do not render in locally generated HTML documentation

gvancuts, 2019-04-05:
The `plus` and `minus` signs that show in the online documentation on the left handside in the navigation panel do not render correctly when generating the HTML version locally. Here is what it looks like:
[image]

gvancuts, 2019-04-05:
@dbkinder , can you take a look at it?

dbkinder, 2019-04-05:
They render fine for me for a local build. Can you look in the _build/html/_static/fonts folder and verify that fontawesome-webfont.woff2 is there.  (That's the font used for the navigation pane markers.)

I'm using a Ubuntu box (as well as WSL on Windows with no issues).  I suspect you're using Clear Linux, right?  I think fontawesome is packaged with the read-the-docs theme so you could try reinstalling that with pip3 (sphinx_rtd_theme==0.4.0)

You can fire up a local server in the _build/html folder  (python3 has one built in) and aim a web browser to http://localhost:8000) and see if there are errors fetching content from your locally generated content.  Here's what I see, (note a 200/OK when fetching fontawesome fonts:
```
~/projectacrn/acrn-hypervisor/doc/_build/html$ python3 -m http.server
Serving HTTP on 0.0.0.0 port 8000 (http://0.0.0.0:8000/) ...
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET / HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET /_static/css/theme.css HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET /_static/pygments.css HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET /_static/acrn-custom.css HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET /_static/js/modernizr.min.js HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET /_static/jquery.js HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:19] "GET /_static/fonts/Lato/lato-regular.woff2 HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/fonts/fontawesome-webfont.woff2?v=4.7.0 HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/fonts/Lato/lato-bold.woff2 HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/fonts/RobotoSlab/roboto-slab-v7-bold.woff2 HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/fonts/RobotoSlab/roboto-slab-v7-regular.woff2 HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/underscore.js HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/doctools.js HTTP/1.1" 200 -
127.0.0.1 - - [05/Apr/2019 09:15:20] "GET /_static/js/theme.js HTTP/1.1" 200 -
```

gvancuts, 2019-04-09:
Hi @dbkinder , thanks for the input. Yes, I'm using Clear Linux (but I'm pretty sure I'm seeing this as well on Fedora). I noticed this morning that this happens **only** if I directly open any other page than the top-level index, i.e.:
```
$ cd acrn-hypervisor/build/doc/html
$ firefox user-guides/acrn-shell.html # the plus/minus sign is not rendered correctly
$ firefix index.hmtl # the plus/minus sign is rendered correctly
```

dbkinder, 2019-04-09:
Ah.  Problem is you're not accessing the content via a server, but via the file system.  If you look at the browser console, you'll see an error: 

Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at file:///home/dbkinder/projectacrn/acrn-hypervisor/doc/_build/html/_static/fonts/fontawesome-webfont.ttf?v=4.7.0. (Reason: CORS request not http). [[Learn More](https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSRequestNotHttp)]

This cross-origin (CORS) request is considered a security issue and so the loading of the fonts is blocked.  Interesting though is that references to the fonts from the home page (or rather from HTML files in the home folder) don't show this error, so I suspect it's the relative references with ``../`` in them that's the problem. 

As you noticed, though that if you start your browser with a reference to index.html in the home folder (``doc/_build/html/index.html``) the font references work.  And if you navigate the doc tree,  the fonts are cached so child pages render as you'd expect.

There's no fix for this if your accessing the content via the file system other than to start at a home page to get the fonts cached before wandering to a page in a subfolder. 

You won't see this problem if you run a web server, for example configure apache to reference your doc/_build/html folder as root, or temporarily run an http server with:

     cd doc/_build/html
     python3 -m http.server &

And then use your browser to open ``localhost:8000``

closed 2019-04-09 by dbkinder

gvancuts, 2019-04-09:
Thanks for the explanation @dbkinder !