projectacrn/acrn-hypervisor #2915: Cannot start Zephyr as UOS with Ubuntu 18.04 as SOS

RoboticsYY, 2019-04-09:
## Environment

* NUC7i7BNH
* Successfully followed [Using Ubuntu as the Service OS](https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html) to set up Ubuntu 18.04 as SOS
* ACRN: acrn-2019w14.4-140000p
* SOS kernel: linux-iot-lts2018-sos-4.19.32-44.x86_64.rpm 

## Problem

I tried to follow [Using Zephyr as User OS](https://projectacrn.github.io/latest/tutorials/usi ng_zephyr_as_uos.html) to setup Zephyr as UOS.

At the second step `Build grub2 boot loader image`, I made some adjustment to pass the build, they are:
* Change "git://git.savannah.gnu.org/gnulib.git" to "https://git.savannah.gnu.org/git/gnulib.git" at file `/zephyrproject/zephyr/boards/x86/common/scripts/grub/src/bootstrap`
* sudo apt install autopoint
* Set ": ${PYTHON:=python3}" in `/zephyrproject/zephyr/boards/x86/common/scripts/grub/src/autogen.sh`

I successfully finished the rest steps of the tutorial, and reboot, then I tried to launch Zephyr as UOS, the screen prints as below:
```
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path ./OVMF.fd, size 0x200000
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/65 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 65 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f568330a000 -> baseaddr 0x0x7f5683400000
mmap 0x8000000@0x7f5683400000
touch 64 pages with pagesz 0x200000
mmap 0x200000@0x7f5783200000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
***********************************************
                                               virt-console backend redirected to /dev/pts/1
                                                                                            ***********************************************
                                                                                                                                           tpm: init_vtpm2:Invalid socket path!
                                                                                                                                                                               polling 33...
                                                                                                                                                                                            Listening 33...
             /bin/sh: 1: /usr/sbin/iasl: not found
                                                  mngr_client_new: Failed to accept from fd 33, err: Invalid argument
                                                                                                                     Stop listening 33...
                                                                                                                                         Stop polling 33...

```

The result of command `dmesg | grep ACRN`:
```
[    0.000000] Hypervisor detected: ACRN
[    0.856392] ACRNTrace: Initialized acrn trace module with 4 cpu
[    0.857841] ACRN HVLog: Failed to init last hvlog devs, errno -19
[    0.858039] ACRN HVLog: Initialized hvlog module with 4 cpu
[   31.844721] ACRN vhm ioeventfd init done!
[   31.844882] ACRN vhm irqfd init done!
```

gvancuts, 2019-04-09:
I hit something (that looks very) similar. There was some sort of a regression in the `OVMF.fd` virtual bootloader, can you try to download the latest version (i.e. the one that was provided by this commit: https://github.com/projectacrn/acrn-hypervisor/commit/296c974dc05904cf019b0b271233613ddaf3c6e3) and try again?

RoboticsYY, 2019-04-09:
@gvancuts I checked out to the commit [296c974](https://github.com/projectacrn/acrn-hypervisor/commit/296c974dc05904cf019b0b271233613ddaf3c6e3), rebuilt and installed ACRN, then copied the new `OVMF.fd`. The outputs are still the same.

gvancuts, 2019-04-09:
Just noticed this error in your output: `/bin/sh: 1: /usr/sbin/iasl: not found`. Make sure you follow these steps on your Ubuntu SOS.
```
sudo apt install acpica-tools
sudo cp /usr/bin/iasl /usr/sbin/iasl
```

RoboticsYY, 2019-04-10:
@gvancuts I have followed these steps, the new output is:
```
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path ./OVMF.fd, size 0x200000
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/65 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 65 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f1d56da3000 -> baseaddr 0x0x7f1d56e00000
mmap 0x8000000@0x7f1d56e00000
touch 64 pages with pagesz 0x200000
mmap 0x200000@0x7f1e56c00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
***********************************************
                                               virt-console backend redirected to /dev/pts/1
                                                                                            ***********************************************
                                             tpm: init_vtpm2:Invalid socket path!
                                                                                 polling 33...
                                                                                             Listening 33...
              /tmp/dm.X45nN2g     30: [0004]		Address : fec00000
                                                                          Error    6302 -  Flag value is too large ^  (Maximum 1 bit)

                                       /tmp/dm.X45nN2g     37: [0004]		Interrupt : 00000002
      Error    6302 -    Flag value is too large ^  (Maximum 1 bit)

                                                                   /tmp/dm.X45nN2g     46: [0004]		Interrupt : 00000009
                                    Error    6302 -    Flag value is too large ^  (Maximum 2 bit)

   /tmp/dm.X45nN2g     47: [0002]		Flags (decoded below) : 000D
                                                                            Error    6302 -                Flag value is too large ^  (Maximum 2 bit)

                                                       mngr_client_new: Failed to accept from fd 33, err: Invalid argument
                            Stop listening 33...
                                                Stop polling 33...

```

gvancuts, 2019-04-10:
This is a problem with the `iasl` tool provided by Ubuntu 18.04, it's at version 20180105 but we need a newer version. Can you try to download and install version 20190405 (from [here](https://www.acpica.org/downloads)). This is what should work with the latest version of ACRN (and v0.8 was just released: https://github.com/projectacrn/acrn-hypervisor/releases/tag/v0.8)

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3383

RoboticsYY, 2019-04-12:
@gvancuts as you suggested, I have made another try with `iasl 20190405`. The new output is:
```
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path ./OVMF.fd, size 0x200000
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/65 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 65 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7fd36849f000 -> baseaddr 0x0x7fd368600000
mmap 0x8000000@0x7fd368600000
touch 64 pages with pagesz 0x200000
mmap 0x200000@0x7fd468400000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
***********************************************
                                               virt-console backend redirected to /dev/pts/2
                                                                                            ***********************************************
                                                                                                                                           tpm: init_vtpm2:Invalid socket path!
                                                                                                                                                                               polling 33...
                                                                                                                                                                                            Listening 33...
             SW_LOAD: partition blob ./OVMF.fd size 2097152 copy to guest 0xffe00000
                                                                                    SW_LOAD: build e820 6 entries to addr: 0x7fd3686ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x0000000007f00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000008000000, size 0x0000000078000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[5]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: ovmf_entry 0xfffffff0
```

After waiting for a few minutes, the screen prints:
```
out instr on NMI port (0x97) not supported
                                          out instr on NMI port (0x97) not supported
                                                                                    out instr on NMI port (0x97) not supported

```
and grub for Zephyr kernel appears. After booting into Zephyr, the output is:
```
WARNING: no console will be available to OS
error: no suitable video mode found.
***** Booting Zephyr OS v1.14.0-rc3-157-g38ecf5b41fe6 *****
Hello World! acrn
```
I think the message means that Zephyr UOS boots successfully and the application works. But I'm not sure if the long time waiting before the grub screen for Zephyr is correct? Another question is how to stop Zephyr UOS?

gvancuts, 2019-04-12:
> out instr on NMI port (0x97) not supported

That I have never seen... @hwang37 , could you ask the engineering team if they know where this could come from?

@RoboticsYY , what is the bios version installed on your NUC?

> and grub for Zephyr kernel appears. After booting into Zephyr, the output is:
>
> ```
> WARNING: no console will be available to OS
> error: no suitable video mode found.
> ***** Booting Zephyr OS [...]

Yes, that means Zephyr is up and running in your VM :-)

There is no mechanism for shutting down this sample app (that I'm aware of) via the console.

hwang37, 2019-04-12:
@KaigeFu to check the Zephyr UOS; @fyin1 to helpUbuntu UOS

RoboticsYY, 2019-04-12:
The bios version on my NUC is `BNKBL357.86A.0076.2019.0219.1646`.

fyin1, 2019-04-12:
@peterfang Suppose this may be related with OVMF.

fuzhongl, 2019-05-05:
Similar issue with: 
[](https://jira.devtools.intel.com/browse/ACRN-3416) Launch Zephyr will take about 5 mins
regression on latest Grub

fuzhongl, 2019-05-30:
@RoboticsYY  The root cause is Zephyr Grub regression.
Please help to close this ticket and check with Zephyr team.
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciated for your understanding.

gvancuts, 2019-05-31:
@fuzhongl , the way GRUB is built and the version that is built hasn't changed in Zephyr since end of March. Do you have any idea what the regression is in this case? How confident are you this in the Grub code? Thanks

fuzhongl, 2019-06-01:
@gvancuts 
Li, Wilson will provide detail about Zephyr GRUB .
Thanks!

fuzhongl, 2019-06-03:
@gvancuts 
Following comments comes from Wilson:
We have a GRUB binary which was built during the middle of March. Using that GRUB binary has no this 5 minutes issue. And we also verified that we can boot into OVMF menu instantly, so the 5-minute issue should come from GRUB.

gvancuts, 2019-06-03:
@fuzhongl , are you able to reproduce this 5-min delay using the latest version of the `build_grub.sh` script?

I'm also wondering if this is specific to using Ubuntu 18.04 as the Service OS, does anyone know?

fuzhongl, 2019-06-03:
> @fuzhongl , are you able to reproduce this 5-min delay using the latest version of the `build_grub.sh` script?
> 
@gvancuts  Yes. please check previous comments: copied as following aslo.

Similar issue with:
[2974](https://github.com/projectacrn/acrn-hypervisor/issues/2974) Launch Zephyr will take about 5 mins
regression on latest Grub

fuzhongl, 2019-06-03:
> I'm also wondering if this is specific to using Ubuntu 18.04 as the Service OS, does anyone know?

This issue isn't related with using Ubuntu 18.04 as the Service OS, for issue [2974](https://github.com/projectacrn/acrn-hypervisor/issues/2974) , it is CL as the Service OS.

gvancuts, 2019-06-04:
> This issue isn't related with using Ubuntu 18.04 as the Service OS, for issue 2974 , it is CL as the Service OS.

OK, let me try to reproduce and investigate this. I was the one who submitted the latest changes to the build_grub.sh script in the Zephyr repo but I did not run into this issue previously.

gvancuts, 2019-06-04:
I have just tested this on a Clear Linux SOS, using a newer version of the script (see https://github.com/zephyrproject-rtos/zephyr/pull/16625) and I can successfully boot Zephyr on ACRN and it's instantaneous.

How did you build Grub (in what environment)?

gvancuts, 2019-06-05:
@RoboticsYY , are you still seeing this issue? I'm not able to reproduce it using Clear Linux as the Service OS (I haven't tried with Ubuntu). If you are still running into this issue, please describe your current environment so I can attempt to reproduce this.

RoboticsYY, 2019-06-05:
@gvancuts I will check my environment, and build with the latest Zephyr, then come back to you.

closed 2020-06-28 by fuzhongl