projectacrn/acrn-hypervisor #2918: UOS fail to resume during suspend/resume testing, SOS is alive, UOS is off, adb disconnected

Mingyuan18, 2019-04-10:
Issue Summary: UOS fail to resume during suspend/resume testing, SOS is alive, UOS is off, adb disconnected, the DUT reboot when press ignition key by manual.

Frequency: 3/10 DUTs,

Preconditons:

1.Wired network support

2.Setup IPATH test env

(1) Download the IPATH installer:

[http://isa-vm.sh.intel.com/IPATH_INSTALLER/IPATH_installer.pyc]

(2) Execute the following command to install IPATH:

python IPATH_installer.pyc

Steps to Reproduce:

1. Connect the device with debug card to test host by USB, start minicom log

2. To IPATH test env, modify test campaign and configure file, and start test

(1) cd to IPATH/testsuite/campaign/kpi$ vi campaign_kpi_BXTP_P_ACRN_WB.xml

(2) cd to IPATH/config/ vim x_device1_execution.config & x_Device_Config_1.py

(3) cd to IPATH/execute, run with commend:

python execute_tests.py ../config/R1J56La006c0f9_device1_execution.config

3. Check the test status during testing

Expect result:

All case test normally,no critical crash happened.

Actual result:

UOS fail to resume during suspend/resume testing, SOS is alive, UOS is off, adb disconnected, the DUT reboot when press ignition key by manual.

 
Additional:

IPATH Link: [http://mtbf-prc38.sh.intel.com/IPATH/IPATH-W10-2019/execute/logs/R1J56L98fb91a1/run_set_2019_03_15_09_52_39/campaign1/test_loop_report.html]

Kernel log: [http://mtbf-prc38.sh.intel.com/IPATH/log/chuankou/acrn-G2018/R1J56L98fb91a1/minicom-SR-3.cap]

                    [http://mtbf-prc38.sh.intel.com/IPATH/log/chuankou/acrn-G2018/R1J56L98fb91a1/minicom-SR-4.cap]

UOS kernel log & dmesg log & aplog: [http://mtbf-prc38.sh.intel.com/IPATH/log/chuankou/acrn-G2018/R1J56L98fb91a1/SR/]

Mingyuan18, 2019-04-10:
[External_System_ID] ACRN-3338

fuzhongl, 2020-05-05:
Fixed with PV version.

closed 2020-05-05 by fuzhongl