projectacrn/acrn-hypervisor #2974: [SIT][ISD][Zephyr] Launch Zephyr will take about 5 mins

Mingyuan18, 2019-04-18:
*Environment*
ClearLinux Version: 28720

*HW/Board*

ISD

*Build link*

N/A

*Steps*

1. Build Zephyr image by follow the guide on https://projectacrn.github.io/latest/tutorials/using_zephyr_as_uos.html?highlight=zephyr
2. Copy the "Zephyr.img", "OVMF.fd" and "launch_zephyr.sh" to ISD 
3. execute the launch_zephyr.sh script

*Expected result*
1. zephyr could boot up successfully and in a short time

*Actual result*
1. zephyr will take about 5 mins to boot up

*Reproduce rate*

Always

*Debugging info*

----

Mingyuan18, 2019-04-18:
[External_System_ID] ACRN-3416

gvancuts, 2019-06-05:
I'm not able to reproduce this on a KBL NUC using the latest code, is this still an issue?

*Note:* you need to use the latest version of Zephyr (that includes this commit https://github.com/zephyrproject-rtos/zephyr/commit/57728ffb276e1c3c6fe0090f7f387a997c605c64) in order to build Grub in a Clear Linux environment.

Mingyuan18, 2019-06-12:
Close with Won't fix.

gvancuts, 2019-06-27:
I still cannot reproduce this with the launch script that we have in the repo, but this is still an issue when you use try to run it as an RTVM. I'll include an example of such a problematic launch script later. In that case, it's known that a different version of Grub works correctly (but nobody has been able to tell yet which exactly :-))

gvancuts, 2019-06-27:
Re-opening

fuzhongl, 2019-07-19:
@gvancuts Any update?
Since this issue isn't ACRN related; we will close it with transferred if no feedback within two weeks.
Thanks for your understanding!

gvancuts, 2019-08-26:
Where would you like to transfer it to?

fuzhongl, 2019-08-27:
@gvancuts Just transfer it to MOSE team, let them know this issue.

closed 2019-10-30 by Mingyuan18