projectacrn/acrn-hypervisor #2984: HV will crash if  launch two UOS with same UUID

lonzoc, 2019-04-19:


Mingyuan18, 2019-04-22:
[External_System_ID] ACRN-3459

fyin1, 2019-04-22:
It's invalid to use same UUID for two different UOS. I believe this is expected behavior.

gvancuts, 2019-04-29:
I don't thing the hypervisor crashing can be considered a valid behaviour, this ought to be fixed in my opinion.

Mingyuan18, 2019-05-09:
@lonzoc , The patch is merge, Please help to confirm the result. Thanks!

Mingyuan18, 2019-05-15:
@lonzoc , We have verified it and the issue has been fixed. Could you please help to confirm it?

fuzhongl, 2019-05-29:
@lonzoc 
Patch was merged in CL 29400, please help to verify this issue; if fixed, please close this issue.
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
You can reopen it if this issue can be reproduced for new release.
Appreciated for your understanding.

Mingyuan18, 2019-06-11:
Close this issue due to no any feedback in two weeks.

closed 2019-06-11 by Mingyuan18