projectacrn/acrn-hypervisor #3032: Dump stack of mem allocation in irq_disabled after using mempool for ACRN VHM

yakuizhao, 2019-04-28:
After the mempool is used for ACRN to fix the VMAP_STACK issue, it will trigger the below issue in course of vhost test.

 1006.317613] BUG: sleeping function called from invalid context at mm/mempool.c:375
[ 1006.318595] in_atomic(): 1, irqs_disabled(): 1, pid: 861, name: vhost-842
[ 1006.319499] Preemption disabled at:
[ 1006.319508] [<ffffffff8829b822>] eventfd_signal+0x22/0x80
[ 1006.321095] CPU: 0 PID: 861 Comm: vhost-842 Tainted: G     U  W         4.19.28-quilt-2e5dc0ac #1
[ 1006.322150] Call Trace:
[ 1006.322887]  dump_stack+0x8e/0xca
[ 1006.323675]  ? eventfd_signal+0x22/0x80
[ 1006.324503]  ___might_sleep.cold.86+0xe6/0xf7
[ 1006.325371]  __might_sleep+0x4a/0x80
[ 1006.326214]  mempool_alloc+0xa7/0x1a0
[ 1006.327071]  ? ___cache_free+0x18a/0x1e0
[ 1006.327956]  acrn_mempool_alloc+0x1f/0x30
[ 1006.328856]  vhm_inject_msi+0x23/0x80
[ 1006.329728]  vhm_irqfd_wakeup+0x3e/0x60
[ 1006.330595]  __wake_up_common+0x74/0x160
[ 1006.331471]  __wake_up_locked_key+0x1b/0x20
[ 1006.332343]  eventfd_signal+0x5c/0x80
[ 1006.333199]  vhost_signal+0x1b7/0x210 [vhost]
[ 1006.334069]  vhost_add_used_and_signal_n+0x2a/0x30 [vhost]
[ 1006.335017]  handle_rx+0xb2/0xb00 [vhost_net]
[ 1006.335933]  ? vhost_worker+0xac/0x130 [vhost]
[ 1006.336862]  ? vhost_worker+0x59/0x130 [vhost]
[ 1006.337791]  handle_rx_kick+0x15/0x20 [vhost_net]
[ 1006.338750]  vhost_worker+0xd5/0x130 [vhost]
[ 1006.339704]  kthread+0x12c/0x150
[ 1006.340618]  ? vhost_dev_init+0x290/0x290 [vhost]
[ 1006.341626]  ? kthread_create_worker_on_cpu+0x70/0x70
[ 1006.342671]  ret_from_fork+0x3a/0x50

yakuizhao, 2019-04-28:
[External_System_ID] ACRN-3476

Mingyuan18, 2019-05-23:
the issue has fixed in daily test:
Installed version: 29360
Daily tag: acrn-2019w19.5-140000p

closed 2019-05-23 by Mingyuan18