projectacrn/acrn-hypervisor #3050: FYI: Kconfiglib major version bumped to 11

ulfalizer, 2019-04-30:
Due to this change, which probably won't affect you: https://github.com/ulfalizer/Kconfiglib/commit/4fed39d9271ceb68be4157ab3f96a45b94f77dc0

Should modify `requirements.txt`, etc., if you want to keep getting the latest version.

Cheers,
Ulf

Mingyuan18, 2019-05-05:
[External_System_ID] ACRN-3622

fuzhongl, 2019-05-14:
It should be fine for "kconfiglib>=10.2" in requirements.txt (under acrn-hypervisor\doc\scripts)

Thanks for reminder.

fuzhongl, 2019-05-29:
@ulfalizer 
Thanks for reminder! Could you please help to close this ticket?
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
Appreciated for your understanding.

Mingyuan18, 2019-06-11:
Close this issue due to no any feedback.

closed 2019-06-11 by Mingyuan18