projectacrn/acrn-hypervisor #3097: USB webcam does not work on UOS over xHCI

kiendinh, 2019-05-10:
I'm using HV (v0.7 and v1.0) on the NUC7i7DNHE connecting to the Logicool C922 USB webcam (uvc cam). The camera is confirmed to work on the SOS using the default Clear Linux desktop app. However, when trying to use the camera on the UOS using xHCI mediator:
-s n,xhci,1:3

the UOS showed errors in dmesg and camera device /dev/video* was not created. The kernel has been configured to support UVC cam already. Depending on the version of ACRN, there were different error message.

With ACRN v0.7 (and acrn-dm), it was:
`usb 1-1: device descriptor read/64, error -110`

With ACRN v1.0, I got
`usb 1-1: device descriptor read/64, error -71`

Thanks

Mingyuan18, 2019-05-10:
This issue is being solved, please reference to the duplicate with #https://github.com/projectacrn/acrn-hypervisor/issues/3054,  thanks!

kiendinh, 2019-05-10:
@Mingyuan18 Thank you. It looks like the issues are not the same. Here the webcam is not even recognized by the UOS. I will check it again when #3054 is solved.

lirui34, 2019-07-15:
[External_System_ID] ACRN-4642

lirui34, 2019-07-15:
Hi @kiendinh, is this issue still reproducible on your NUC with the latest hypervisor ?

Mingyuan18, 2019-08-08:
@kiendinh ,Patch merged on ww29.1 . Please help to double check the issue result. Thanks!

fuzhongl, 2020-06-28:
@kiendinh  Please reopen this git issue if you still met same issue with latest code.
Thanks!

closed 2021-03-15 by fuzhongl