projectacrn/acrn-hypervisor #3126: Documentation: "Using Partition mode on UP2" tutorial is outdated

gvancuts, 2019-05-15:
The ["Using Partition mode on UP2" tutorial](https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_up2.html) is outdated and needs to refreshed. There are references to files that are no longer in use such as `pt_dev.c` and `partition_config.h` (and probably more than that).

gvancuts, 2019-05-29:
@NanlinXie , can you assign someone to this?

lirui34, 2019-07-15:
Hi, @gvancuts  Can you please close this issue as the document is already published to ``v1.1`` ?

gvancuts, 2019-09-03:
Hi @lirui34 , the PR referenced here (#3113) introduced a **new** tutorial for a different platform. The tutorial pertaining to the UP2 board is still part of our documentation (and still outdated).

gvancuts, 2020-01-10:
Can someone take a look at this? I see some updates were made recently (in December) but we are still referencing v0.6 for ACRN... my recommendation is that we either update this tutorial to use the latest version of ACRN (1.5) or remove it altogether. My preference is that we do the former.

fuzhongl, 2020-05-04:
@gvancuts Following Doc is updated to v1.6:
https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_nuc.html
Can we close this git issue?
Thanks!

gvancuts, 2020-05-06:
Yes, thanks!

closed 2020-05-06 by gvancuts