projectacrn/acrn-hypervisor #3138: DM: Decouple and increase kernel boot args length

vijaydhanraj, 2019-05-18:
Currently, we use STR_LEN for all checking the size of all the acrn-dm parameters. But some parameters like kernel boot args can grow based on different needs. For example, when kata launches guest VM using acrn, the kernel boot args increases by 256 bytes (i.e 1024 +256). 

Just increasing STR_LEN will unnecessarily increase allocations for other acrn-dm parameters. So decoupling only boot_args length and increasing it to 2048.  

PS: If other parameters like ramdisk path, kernel path, elf_path etc. don't need 1024 bytes, we can reduce this to 256 or 512 bytes.

Mingyuan18, 2019-05-20:
[External_System_ID] ACRN-3823

Mingyuan18, 2019-06-12:
Can't reproduce this issue in below build:
ACRN tag: acrn-2019w22.1-140000p
CL: 29660

closed 2019-06-12 by Mingyuan18