projectacrn/acrn-hypervisor #3166: hv: the boot issue of sharing one MSR bitmap per VM for X2APIC APICV mode switch

ZideChen0, 2019-05-23:
It could have synchronous issues when multiple vcpu in one guest to switch to x2apic mode.

because msr_bitmap is per vm, so for example:

update_msr_bitmap_x2apic_apicv{
   // vcpu0 has call the following function, which to enable intercepting MSR_IA32_EXT_APIC_EOI
   intercept_x2apic_msrs(msr_bitmap, INTERCEPT_WRITE);

  // vcpu1 writes to EOI and it's intercepted by hypervisor, which doesn't handle this MSR and the vmexit results in a failure.
    enable_msr_interception(msr_bitmap, MSR_IA32_EXT_APIC_EOI, INTERCEPT_DISABLE);
}

We may either make the msr_bitmap per-vcpu, which is simple, clean, but take 4KB per vcpu;
or we can carefully write the update_msr_bitmap_x2apic_xx() functions to avoid the issue, but the solution is not clean.

ZideChen0, 2019-05-23:
[External_System_ID] ACRN-3867

Mingyuan18, 2019-06-24:
Patch merged in CL 29660 
acrn tag: acrn-2019w21.5-140000p

closed 2019-06-24 by Mingyuan18