projectacrn/acrn-hypervisor #3189: DM: handle virtio-console writes when no socket backend is connected

vijaydhanraj, 2019-05-29:
When virtio-console is used as console port with socket backend, guest kernel tries to hook it up with hvc console and sets it up. It doesn't check if a client is connected and can result in ENOTCONN with virtio-console backend being reset. This will prevent client connection at a later point. To avoid this, ignore ENOTCONN error.

PS: For Kata, the runtime first launches VM and then proxy which acts as a client connects to this socket. If this error is not handled, proxy will never be able to connect as the backend itself will be reset.

vijaydhanraj, 2019-05-29:
[External_System_ID] ACRN-3796

Mingyuan18, 2019-07-19:
The patch was merged and no regression issue in daily test.

closed 2019-07-19 by Mingyuan18