projectacrn/acrn-hypervisor #3198: Enabling acrn on qemu.

Mingyuan18, 2019-05-30:
Enabling acrn on qemu.

Mingyuan18, 2019-05-30:
[External_System_ID] ACRN-3928

fuzhongl, 2019-06-26:
@Mingyuan18  The patch was merged into (tag: acrn-2019w22.5-140000p) CL 29790.
Could you please close this ticket?
Thanks!

gvancuts, 2019-06-26:
Can you also plan to add some documentation on how to run ACRN on QEMU?

tw4452852, 2019-06-27:
@fuzhongl There's still one patch under review. So just hold on.

tw4452852, 2019-06-27:
@gvancuts I will add a guild once all the code changes have been merged.

LiangZhou9527, 2019-10-02:
just curious:  is there any guide about how to run acrn on qemu?

gvancuts, 2019-10-02:
Hi @LiangZhou9527 , I haven't seen one yet. FYI, we may not get any better answer since most of the engineering team is in China and on holiday this week.

yfliuu, 2020-01-08:
Hi @tw4452852  @gvancuts, is there any tutorial on running ACRN on QEMU/KVM? I've tried but ACRN complains about CPUID level being too small (13). Manually setting it to 22 (the value on physical machine) won't work as the hypervisor will crash.

gvancuts, 2020-01-10:
I haven't seen one yet. @tw4452852 @NanlinXie @fuzhongl , do you know if we have instructions on how to run ACRN on QEMU? If so, it would be good to make a tutorial out of those!

shlomopongartz, 2020-03-10:
Any updates on how to run ACRN on QEMU/KVM?

NanlinXie, 2020-03-12:
Hi, Pongratz,

         We have plan to support it in ACRN v2.0 release(~2020.6), and will publish BKM once we done.  Please take care of our Release note for update. Thanks for your interest on ACRN.

Best wishes!
Nanlin

gvancuts, 2020-06-03:
@shlomopongartz , you can now run ACRN on QEMU/KVM. The tutorial has been published here: https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html

Can you close this issue if this satisfies your needs?

shlomopongartz, 2020-06-04:
Hi,
Thank you.
Why do you want me to close the issue? I didn't open it.
I just asked a question, I see that the one who open it is Mingyuan18  (see on top).
Best regards,
S.P.

gvancuts, 2020-06-04:
> Why do you want me to close the issue? I didn't open it.

Indeed... no action needed your part... my bad! :-)

fuzhongl, 2020-06-28:
Integrated into acrn v2.0

closed 2020-06-28 by fuzhongl