projectacrn/acrn-hypervisor #3200: acrn-crashlog: error logs in client write to server via socket

Mingyuan18, 2019-05-30:
Summary:
acrn-crashlog: error logs in client write to server via socket
Test steps:
1. trigger a coredump
2. journalctl to check usercrash logs
Expected result:
After step 2, there isn't usercrash error logs displayed
Actual result:
After step 2, error logs as below:
Oct 09 18:53:21 clr-9c8b2eab63d64d0f940ba69c3cce3d68 usercrash_s[260]: crash socket received short read of length 10,
Oct 09 18:53:21 clr-9c8b2eab63d64d0f940ba69c3cce3d68 usercrash_s[260]: (expected 72)
Oct 09 18:53:21 clr-9c8b2eab63d64d0f940ba69c3cce3d68 usercrash_c[463]: failed to notify usercrashd of completion

Mingyuan18, 2019-05-30:
[External_System_ID] ACRN-3932

fuzhongl, 2019-06-26:
@Mingyuan18  Patch merged into (tag: acrn-2019w22.5-140000p) CL 29790.
Could you please close this ticket?
Thanks!

Mingyuan18, 2019-07-01:
No regression found in daily test. So close this issue.

closed 2019-07-01 by Mingyuan18