projectacrn/acrn-hypervisor #3224: Build failure with acrn 1.0

rossburton, 2019-06-04:
We hit an intermittant build failure in the hypervisor:

```
/data/chinhuat/git/yocto/build/meta-acrn/tmp-sos/hosttools/bash scripts/genld.sh bsp/ld/link_ram.ld.in WORKDIR/build//hypervisor/link_ram.ld WORKDIR/build//hypervisor/.config
x86_64-oe-linux-gcc  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=WORKDIR/recipe-sysroot -Wl,-Map=WORKDIR/build//hypervisor/acrn.map -o WORKDIR/build//hypervisor/acrn.out  -Wl,--gc-sections -nostartfiles -nostdlib -Wl,-n,-z,max-page-size=0x1000 -pie -z noreloc-overflow   -TWORKDIR/build//hypervisor/link_ram.ld WORKDIR/build//hypervisor/boot/reloc.o WORKDIR/build//hypervisor/arch/x86/init.o WORKDIR/build//hypervisor/arch/x86/seed/seed.o WORKDIR/build//hypervisor/arch/x86/seed/seed_abl.o WORKDIR/build//hypervisor/arch/x86/seed/seed_sbl.o WORKDIR/build//hypervisor/arch/x86/configs/vm_config.o WORKDIR/build//hypervisor/arch/x86/configs/nuc7i7bnh/ve820.o WORKDIR/build//hypervisor/arch/x86/configs/nuc7i7bnh/board.o WORKDIR/build//hypervisor/scenarios/sdc/vm_configurations.o WORKDIR/build//hypervisor/boot/acpi.o WORKDIR/build//hypervisor/boot/dmar_parse.o WORKDIR/build//hypervisor/arch/x86/ioapic.o WORKDIR/build//hypervisor/arch/x86/lapic.o WORKDIR/build//hypervisor/arch/x86/cpu.o WORKDIR/build//hypervisor/arch/x86/cpu_caps.o WORKDIR/build//hypervisor/arch/x86/security.o WORKDIR/build//hypervisor/arch/x86/mmu.o WORKDIR/build//hypervisor/arch/x86/e820.o WORKDIR/build//hypervisor/arch/x86/pagetable.o WORKDIR/build//hypervisor/arch/x86/page.o WORKDIR/build//hypervisor/arch/x86/notify.o WORKDIR/build//hypervisor/arch/x86/vtd.o WORKDIR/build//hypervisor/arch/x86/gdt.o WORKDIR/build//hypervisor/arch/x86/irq.o WORKDIR/build//hypervisor/arch/x86/timer.o WORKDIR/build//hypervisor/arch/x86/vmx.o WORKDIR/build//hypervisor/arch/x86/cpu_state_tbl.o WORKDIR/build//hypervisor/arch/x86/pm.o WORKDIR/build//hypervisor/arch/x86/static_checks.o WORKDIR/build//hypervisor/arch/x86/trampoline.o WORKDIR/build//hypervisor/arch/x86/guest/vcpuid.o WORKDIR/build//hypervisor/arch/x86/guest/vcpu.o WORKDIR/build//hypervisor/arch/x86/guest/vm.o WORKDIR/build//hypervisor/arch/x86/guest/vlapic.o WORKDIR/build//hypervisor/arch/x86/guest/vmtrr.o WORKDIR/build//hypervisor/arch/x86/guest/guest_memory.o WORKDIR/build//hypervisor/arch/x86/guest/vmcall.o WORKDIR/build//hypervisor/arch/x86/guest/vmsr.o WORKDIR/build//hypervisor/arch/x86/guest/instr_emul.o WORKDIR/build//hypervisor/arch/x86/guest/ucode.o WORKDIR/build//hypervisor/arch/x86/guest/pm.o WORKDIR/build//hypervisor/arch/x86/guest/assign.o WORKDIR/build//hypervisor/arch/x86/guest/ept.o WORKDIR/build//hypervisor/arch/x86/guest/vmx_io.o WORKDIR/build//hypervisor/arch/x86/guest/virq.o WORKDIR/build//hypervisor/arch/x86/guest/virtual_cr.o WORKDIR/build//hypervisor/arch/x86/guest/vmcs.o WORKDIR/build//hypervisor/arch/x86/guest/vmexit.o WORKDIR/build//hypervisor/arch/x86/guest/trusty.o WORKDIR/build//hypervisor/arch/x86/cat.o WORKDIR/build//hypervisor/lib/string.o WORKDIR/build//hypervisor/lib/memory.o WORKDIR/build//hypervisor/lib/crypto/crypto_api.o WORKDIR/build//hypervisor/lib/crypto/mbedtls/hkdf.o WORKDIR/build//hypervisor/lib/crypto/mbedtls/sha256.o WORKDIR/build//hypervisor/lib/crypto/mbedtls/md.o WORKDIR/build//hypervisor/lib/crypto/mbedtls/md_wrap.o WORKDIR/build//hypervisor/lib/sprintf.o WORKDIR/build//hypervisor/common/softirq.o WORKDIR/build//hypervisor/common/hv_main.o WORKDIR/build//hypervisor/common/hypercall.o WORKDIR/build//hypervisor/common/trusty_hypercall.o WORKDIR/build//hypervisor/common/schedule.o WORKDIR/build//hypervisor/common/vm_load.o WORKDIR/build//hypervisor/common/ptdev.o WORKDIR/build//hypervisor/common/static_checks.o WORKDIR/build//hypervisor/common/stack_protector.o WORKDIR/build//hypervisor/hw/pci.o WORKDIR/build//hypervisor/dm/vpic.o WORKDIR/build//hypervisor/dm/vrtc.o WORKDIR/build//hypervisor/dm/vioapic.o WORKDIR/build//hypervisor/dm/io_req.o WORKDIR/build//hypervisor/dm/vpci/vdev.o WORKDIR/build//hypervisor/dm/vpci/vpci.o WORKDIR/build//hypervisor/dm/vmptable.o WORKDIR/build//hypervisor/dm/vpci/vhostbridge.o WORKDIR/build//hypervisor/dm/vpci/pci_pt.o WORKDIR/build//hypervisor/dm/vpci/vmsi.o WORKDIR/build//hypervisor/dm/vpci/vmsix.o WORKDIR/build//hypervisor/bsp/firmware_wrapper.o WORKDIR/build//hypervisor/bsp/firmware_sbl.o WORKDIR/build//hypervisor/bsp/firmware_uefi.o WORKDIR/build//hypervisor/bsp/cmdline.o WORKDIR/build//hypervisor/bsp/const_dmar.o WORKDIR/build//hypervisor/boot/uefi/uefi_boot.o WORKDIR/build//hypervisor/boot/sbl/multiboot.o WORKDIR/build//hypervisor/arch/x86/boot/cpu_primary.o WORKDIR/build//hypervisor/arch/x86/boot/cpu_save_boot_ctx.o WORKDIR/build//hypervisor/arch/x86/boot/trampoline.o WORKDIR/build//hypervisor/arch/x86/idt.o WORKDIR/build//hypervisor/arch/x86/wakeup.o WORKDIR/build//hypervisor/arch/x86/sched.o WORKDIR/build//hypervisor/arch/x86/guest/vmx_asm.o WORKDIR/build//hypervisor/arch/x86/retpoline-thunk.o WORKDIR/build//hypervisor/debug/libdebug.a
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/build//hypervisor/arch/x86/init.o: in function `init_debug_post':
WORKDIR/git/hypervisor/arch/x86/init.c:49: undefined reference to `profiling_setup'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/build//hypervisor/common/hv_main.o: in function `vcpu_thread':
WORKDIR/git/hypervisor/common/hv_main.c:50: undefined reference to `profiling_vmenter_handler'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/common/hv_main.c:64: undefined reference to `profiling_pre_vmexit_handler'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/common/hv_main.c:78: undefined reference to `profiling_post_vmexit_handler'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/build//hypervisor/debug/libdebug.a(hypercall.o): in function `hcall_profiling_ops':
WORKDIR/git/hypervisor/debug//hypercall.c:33: undefined reference to `profiling_msr_ops_all_cpus'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:36: undefined reference to `profiling_vm_list_info'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:39: undefined reference to `profiling_get_version_info'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:42: undefined reference to `profiling_get_control'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:45: undefined reference to `profiling_set_control'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:48: undefined reference to `profiling_configure_pmi'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:51: undefined reference to `profiling_configure_vmsw'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:54: undefined reference to `profiling_get_pcpu_id'
WORKDIR/recipe-sysroot-native/usr/bin/x86_64-oe-linux/../../libexec/x86_64-oe-linux/gcc/x86_64-oe-linux/9.1.0/ld: WORKDIR/git/hypervisor/debug//hypercall.c:57: undefined reference to `profiling_get_status_info'
collect2: error: ld returned 1 exit status
Makefile:350: recipe for target 'WORKDIR/build//hypervisor/acrn.out' failed
make: *** [WORKDIR/build//hypervisor/acrn.out] Error 1
make: Leaving directory 'WORKDIR/git/hypervisor'
```

Rebuilding worked, so I suspect this is a build race.

gvancuts, 2019-06-11:
FWIW, I have also seen that when building on Clear Linux (so this does not seem to be Yocto-specific).

Mingyuan18, 2019-06-12:
[External_System_ID] ACRN-4038

fuzhongl, 2019-06-13:
No this issue for our CI build system.
@rossburton  Could you please share the command you used for make hypervisor?
Thanks!

rossburton, 2019-06-17:
```
make -C hypervisor
```

It looks like a race so the window may well be very small.  I've never seen it myself, but a colleague did.

fuzhongl, 2020-05-05:
@rossburton 
The race issue about multi-core build system is fixed recently:
https://github.com/projectacrn/acrn-hypervisor/pull/4619
Do your colleague still met this race issue with the latest codebase? 
Thanks!

rossburton, 2020-05-05:
CCing @saininav as I don't work on acrn anymore.

fuzhongl, 2020-06-28:
race issue is fixed in ACRN 1.6.1

closed 2020-06-28 by fuzhongl