projectacrn/acrn-hypervisor #3226: OVMF release v1.1

peterfang, 2019-06-04:
OVMF release v1.1
- Cleanup unused and unsupported function of AcrnRtc
- Remove SMBIOS support
- Add ResetSystemLib for ACRN
- Disable BhyveFwCtlLib
- Disable COM2
- Avoid connecting all system drivers
- Speed up Ps2KeyboardDxe
- Created new Rtc driver and optimized RtcInit and RtcGetTime
- Initialize EmuVariable NV storage memory with varstore from ROM
- Switch to BasePciLibPciExpress
- Avoid exhaustive PCI bus scanning during PCI host bridge initialization
- Add support for ACRN platform initialization
- Add ACRN's Host Bridge Device ID

peterfang, 2019-06-04:
[External_System_ID] ACRN-3963

Mingyuan18, 2019-07-19:
The patch was merged and V1.1 was relese.

closed 2019-07-19 by Mingyuan18