projectacrn/acrn-hypervisor #3231: devicemodel fails to build

shrmrf, 2019-06-05:
Following instructions on Ubuntu 16.04, I am able to successfully build the hypervisor (`make hypervisor`) and tools (`make tools`). But, `make devicemodel` fails with the following errors:

```console
[ ! -e /disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/devicemodel/hw/pci/virtio/virtio_gpio.o ] && mkdir -p /disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/devicemodel/hw/pci/virtio/; \
cc -g -O0 -std=gnu11 -D_GNU_SOURCE -DNO_OPENSSL -m64 -Wall -ffunction-sections -Werror -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -fpie -Wno-stringop-truncation -Wno-address-of-packed-member -I/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/devicemodel/include -I/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/devicemodel/include/public -I/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/devicemodel/include -I/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/tools -fstack-protector-strong -DDM_DEBUG -c hw/pci/virtio/virtio_gpio.c -o /disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/devicemodel/hw/pci/virtio/virtio_gpio.o
hw/pci/virtio/virtio_gpio.c:20:10: fatal error: linux/gpio.h: No such file or directory
 #include <linux/gpio.h>
          ^~~~~~~~~~~~~~
compilation terminated.
Makefile:216: recipe for target '/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/devicemodel/hw/pci/virtio/virtio_gpio.o' failed
make[1]: *** [/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/build/devicemodel/hw/pci/virtio/virtio_gpio.o] Error 1
make[1]: Leaving directory '/disk1/localdisk/home/taimoor/dev/github/kubernetes/binutils-2.27/acrn-hypervisor/devicemodel'
Makefile:46: recipe for target 'devicemodel' failed
make: *** [devicemodel] Error 2
```
[Complete log file is here :wave:](https://github.com/projectacrn/acrn-hypervisor/files/3257121/makelog.log)

shrmrf, 2019-06-05:
To get a standardized build, I did:
```bash
docker build -t ubuntu-acrn -f .travis-dockerfiles/Dockerfile.ubuntu16.04 .
docker run -v `pwd`:/root/acrn ubuntu-acrn /bin/bash -c "make clean && make"
```

[Log file here :wave:](https://github.com/projectacrn/acrn-hypervisor/files/3257201/makelog.log)

shrmrf, 2019-06-05:
Can someone point me to the Travis logs? I can't find this project any longer on Travis' website.

@gvancuts you seem to be the author of the `Dockerfile`s.. any idea?

gvancuts, 2019-06-06:
Hi @shrmrf , we have never enabled Travis on this repo. Instead we use a Jenkins-based CI system. As a result, these Dockerfiles are largely unmaintained, I should really remove them since we are not using them at all in the project. They are still useful to me occasionally but I could just as well keep those in a dedicated repo somewhere else. I updated the Ubuntu 16.04 last in April (commit) but was seeing this compilation error already. I never dug further to see how I could resolve this (I don't think many people use Ubuntu 16.04 anymore).

Are you tied to using Ubuntu 16.04 for this or could you use a container based on a different OS?

shrmrf, 2019-06-06:
@gvancuts enabling travis/azure/circle etc. as well might actually be nice because then the build logs are open to the public. If you're using Jenkins, a `Jenkinsfile` can also be placed I guess instead of the [`.travis.yml` file](https://github.com/projectacrn/acrn-hypervisor/blob/master/.travis.yml)

My workstation is 16.04 but I'm free to use any other OS (via `docker`) to do the compilation etc.
I was under the impression that Ubuntu 16.04 is still a "supported" distro. The docs still have instructions in that regard. That's why I reported the bug here.

gvancuts, 2019-06-06:
I agree that Travis or other public CI system would be nice (which is why I started developing these Dockerfiles in the first place :-)). I'm not sure how the Jenkins CI is set-up to be honest. Maybe @NanlinXie knows more about it.

You're right. we've never officially removed Ubuntu 16.04 from the list. It has become a little more cumbersome to use it with time as we need to manually install versions of gcc, binutils and the GPIO headers that are not part of their repos. It's also broken at the moment as you can see. The problem comes from the `gpio.h` file, that in the Docker container, we get from a package called `openwince-include`. That header file is not correct. I replaced it with one from my Fedora 29 workstation (`/usr/include/linux/gpio.h`) and after that I was able to compile ACRN. I'm not sure what a better source for getting that file in Ubuntu 16.04 would be, if you have ideas, let me know and we can patch the `Dockerfile` and update the instructions.

kriskep, 2019-06-07:
> That header file is not correct. I replaced it with one from my Fedora 29 workstation (`/usr/include/linux/gpio.h`) and after that I was able to compile ACRN. I'm not sure what a better source for [...]

@gvancuts on Ubuntu 16.04, the file can be found in `/usr/src/linux-headers-[kernelversion]/include/uapi/linux/gpio.h`

shrmrf, 2019-06-07:
👍 for travis or any public CI/CD!!

gvancuts, 2019-06-07:
> @gvancuts on Ubuntu 16.04, the file can be found in `/usr/src/linux-headers-[kernelversion]/include/uapi/linux/gpio.h`

Thanks @kriskep . Do you know which package installs that file? I installed `linux-headers-generic` but that did not install it?

fuzhongl, 2019-06-10:
@gvancuts As mentioned at [2737](https://github.com/projectacrn/acrn-hypervisor/issues/2737)Build ACRN failed in Ubuntu 16.04

run these command:
apt-get install openwince-include 
cp /usr/include/openwince/arm/sa11x0/gpio.h  /usr/include/linux/

gvancuts, 2019-06-10:
> @gvancuts As mentioned at [2737](https://github.com/projectacrn/acrn-hypervisor/issues/2737)Build ACRN failed in Ubuntu 16.04
> 
> run these command:
> apt-get install openwince-include
> cp /usr/include/openwince/arm/sa11x0/gpio.h /usr/include/linux/

@fuzhongl , as mentionned above https://github.com/projectacrn/acrn-hypervisor/issues/3231#issuecomment-499645976, this is no longer working. I tried to use the `gpio.h` file that comes from my Fedora 29 workstation (which comes from the `kernel-headers` package) and that allowed me to build ACRN (I have not tested the binaries though).

@kriskep mentinoned to me (offline) that even though it's possible to build ACRN in Ubuntu 16.04 with this workaround, the binary fails to work. I'm thinking it may be time to stop our support for Ubuntu 16.04 as a development machine. It's easy enough to build a container based on a different distro to work around that for those who still use Ubuntu 16.04.

fuzhongl, 2019-06-10:
I'm thinking it may be time to stop our support for Ubuntu 16.04 as a development machine. It's easy enough to build a container based on a different distro to work around that for those who still use Ubuntu 16.04.

@gvancuts  Totally agree!

lirui34, 2019-07-15:
Hi, @shrmrf  
Can we now close this issue as it's not recommended to build ``devicemodel``on Ubuntu 16.04 ?

shrmrf, 2019-07-15:
Yeah, sure... I just raised the issue to let you guys know

closed 2019-07-16 by lirui34