projectacrn/acrn-hypervisor #3249: handle the emulation PCI cfg access from unaligned reg_offset

yakuizhao, 2019-06-12:
When the SOS kernel/pre-launched OS access the 0xCF8/0xCFC, it will cause
the vm-exit and then the hypervisor tries to emulate the PCI_cfg access.

If the aligned reg is passed in 0xCF8 register, it can work well. But when
the unaligned reg is passed in 0xCF8 register, the cached_reg + offset will cause
that the incorrect pci_cfg offset is accessed. For example:
   The cached_reg = 0x02(Device_ID offset) based on the value passed from 0xCF8
   offset = 2 based on 0xCFC-0xCFF address.
   Then cached_reg + offset is used as the offset(PCI_CMD_REG)
 
In fact the unaligned reg can work well on the real HW.

yakuizhao, 2019-06-13:
[External_System_ID] ACRN-4037

Mingyuan18, 2019-09-26:
The patch was merged and no regression in this tag:
tag:acrn-2019w25.4-140000p

closed 2019-09-26 by Mingyuan18