projectacrn/acrn-hypervisor #3272: Can't launch 2nd VMs

ttzeng, 2019-06-17:
We fail to launch the 2nd Clear Linux VMs using the ACRN hypervisor ``acrn.efi`` after the tag ``acrn-2019w15.1-140000p``. 

The 2nd VM uses a dedicated Clear Linux KVM image, a new ``tap1`` net device. The scenario works fine using Clear Linux 28790 as SOS, but the DM shipped with recent Clear Linux releases always display "failed to create VM vm2" while launching the 2nd VM. We can launch the 2nd VM with the later Clear Linux release (e.g. CL 29990) only after rolling back the ``acrn.efi`` to the code built from the tag ``acrn-2019w15.1-140000p``

Is it a regression? or the 2nd ``launch_uos.sh`` script need to be modified to provide certain DM arguments?

Mingyuan18, 2019-06-18:
[External_System_ID] ACRN-4096

fuzhongl, 2019-06-18:
@ttzeng For SDC config, It can't launch 2nd VMs with newer release.; It works as design.
Please help to close this ticket.
Thanks!

fuzhongl, 2019-06-18:
@ttzeng For SDC config, it can't launch 2nd VMs with newer release, works as design.
Please help to close this ticket.
Thanks！

ttzeng, 2019-06-18:
@fuzhongl thanks for the clarification. If it's by design that the current ACRN code does not support 2 LaaG instances, we may need to document somewhere to remind users this constraint. How do you think?

closed 2019-06-18 by ttzeng

gvancuts, 2019-06-18:
> @ttzeng For SDC config, it can't launch 2nd VMs with newer release, works as design.
I agree with @ttzeng , we really need to document this design. The only reference I could find to it is via the `Kconfig` symbol table... that's really terse and really does not explain much at all. There is no other reference in our Developer documentation/HLDs. I'll file a ticket to track this.