projectacrn/acrn-hypervisor #3283: New scenario-based configurations lack documentation

gvancuts, 2019-06-18:
A new "scenario-based" configuration mechanism was recently introduced in ACRN, this is selected via the `Kconfig` configuration tool of the hypervisor. There is no existing documentation beyond the rather terse explanation in the `Kconfig` symbol table. We should provide additional documentations that cover the following:
- More complete description of the different scenario
- How does one change/add a new scenario (only if customers are supposed to do this)

The user experience is seriously affected by this as there are things that are no longer possible without reconfiguring and recompiling the hypervisor - see also #3272 as an example.

NanlinXie, 2019-06-19:
I just discussed with @dbkinder that we provide short FAQ to describe the solution for #3272. Please let me if you think more complete description is better. @gvancuts @dbkinder

gvancuts, 2019-06-19:
I think we need more than a FAQ but this is already a good first step.

Mingyuan18, 2019-06-26:
[External_System_ID] ACRN-4253

Mingyuan18, 2019-08-12:
@gvancuts , Doc PR for SDC2 and Industry secnario merged. Please help to confirm. Thanks!

gvancuts, 2019-08-19:
I haven't gone through all the new documentation yet but this is looking good, closing this. Thanks!

closed 2019-08-19 by gvancuts