projectacrn/acrn-hypervisor #3291: Update documentation and helper scripts to use newer `swupd` commands

gvancuts, 2019-06-19:
A number of `swupd` commands are being deprecated in favour of new ones (all existing functionalities have been preserved). The specific changes are:
- `swupd diagnose` (that replaces `verify`): https://github.com/clearlinux/swupd-client/commit/2cb0dd80b4ba98b63c754cb5562c297f0cebdc1a
- `--version` flag (instead of `-m`): https://github.com/clearlinux/swupd-client/commit/a3131db0feb70f4a7616d29e38742d8d7a81c984
- `os-install` (that replaces `verify --install`): https://github.com/clearlinux/swupd-client/commit/a84c60a500ca6efe76368919542a11ee01de6184

A few documents and helper scripts need to be updated accordingly:
* doc/getting-started/acrn_quick_setup.sh
* doc/getting-started/apl-nuc.rst
* doc/tutorials/create-up2-images.sh
* doc/tutorials/skl-nuc.rst
* doc/tutorials/agl-vms.rst

Note that we need to pay attention to the fact the new instructions/scripts will need to be used with a version of Clear Linux that has a `swupd` client with these new commands.

Mingyuan18, 2019-06-20:
[External_System_ID] ACRN-4151

Mingyuan18, 2019-07-24:
@gvancuts  Do you want to update another patch?

fuzhongl, 2020-05-05:
@gvancuts 
acrn_quick_setup.sh was updated.
Can this git issue be closed now?
Thanks!

gvancuts, 2020-05-05:
Yes, closing it.

closed 2020-05-05 by gvancuts