projectacrn/acrn-hypervisor #3357: DM: support virtio-i2c

conghuic23, 2019-07-03:
In current code, when the Guest OS wants to access to a slave I2C device, it should pass through the whole I2C adapter. But if the adapter has two or more slave devices, and need to be used by different Guests, pass through cannot meet the requirement.

To be more flexible, virtio-i2cadap was put forward, and it should have features:
 1. Work as an I2C adapter.
 2. Slave devices under same physical adapter can be assigned to  different virtio-i2c adapters.
 3. Support >= 1 slave devices.
 4. I2C slave devices driver needn't change.

Mingyuan18, 2019-07-04:
[External_System_ID] ACRN-2460

Mingyuan18, 2019-07-24:
The patch was merged and no regression daily test
ACRN TAG: acrn-2019w28.1-140000p
CL: 30210

closed 2019-07-24 by Mingyuan18