projectacrn/acrn-hypervisor #3378: UOS can't get IP address after docker is enabled

ttzeng, 2019-07-08:
The UOS can't get IP address after starting the docker in SOS. The virtio-net device works fine in the UOS before starting the docker in SOS. Is this a bug or a constraint?

Here are steps to replicate the issue:
(1) Set up SOS + UOS by following the GSG
(2) Verify the UOS network functions normally before shutdown the UOS
(3) `sudo swupd bundle-add docker-compose` in SOS if no docker installed before
(4) `sudo systemctl start docker` to start the docker service
(5) Launch the UOS again, this time the network device is there, but no IP address is assigned.

lirui34, 2019-07-09:
[External_System_ID] ACRN-4597

fuzhongl, 2019-08-20:
@ttzeng Please have a try with following steps:
modify /usr/lib/systemd/system/docker.service：

       ExecStart add “--ip-forward=false”

then systemctl start docker.service                   

       UOS will get IP address

ttzeng, 2019-08-20:
@fuzhongl it works! Thanks for the cure. :)

closed 2019-08-20 by ttzeng