projectacrn/acrn-hypervisor #3379: DM: Increase hugetlbfs MAX_PATH_LEN from 128 to 256.

vijaydhanraj, 2019-07-08:
Current hugetlbfs code has a limitation on file path length. The path string comprises of mount path + vm name. Something like /run/hugepage/acrn/huge_lv1/vm1. To this UUID (32 bytes) is added and the total path length should be less than 128.
This works fine but in cases where the VM name is large as in case kata, this check fails. Kata passes a sandbox-id as VM name and so path + 32 for UUID easily exceed 128 bytes.
“/run/hugepage/acrn/huge_lv1/sandbox-6d455fa48788eae82dee42410fc3d38849c2a5196f930b3d6944805aed8d24c7

To address this, increase the size of MAX_PATH_LEN from 128 to 256 bytes.

lirui34, 2019-07-09:
[External_System_ID] ACRN-4598

Mingyuan18, 2019-07-19:
No regression issue found in below build:
ACRN tag: acrn-2019w28.3-140000p
CL : 30340

closed 2019-07-19 by Mingyuan18