projectacrn/acrn-hypervisor #3406: Variable Argument List Read Without Ending with va_end

lirui34, 2019-07-12:
va_end was not getting called under certain condition after the va_start function call taking the valist. Variable argument list to a function if not defined appropriately can be exploited by an attacker to perform malicious stack operations and induce of denial of service.
Below is the location of the code, where this issue was found:

acrn-hypervisor\tools\acrn-crashlog\usercrash\crash_dump.c | loginfo()
acrn-hypervisor\tools\acrn-crashlog\common/log_sys.c| debug_log().
acrn-hypervisor\devicemodel/hw/platform/acpi/acpi.c| dsdt_line()

lirui34, 2019-07-12:
[External_System_ID] ACRN-4550

Mingyuan18, 2019-07-24:
The patch was merged and pre-integration test no regression issue found:
ACRN TAG: acrn-2019w29.4-150000p
CL: 30400

closed 2019-07-24 by Mingyuan18