projectacrn/acrn-hypervisor #3407: Buffer Overflow – Array Index Out of Bounds

lirui34, 2019-07-12:
We analysed that a buffer overflow condition was detected that could reach out of bounds for of the index of the array object in the <function> function.
acrn-hypervisor/hypervisor/common/hypercall.c:441 | inject_msi_lapic_pt()
acrn-hypervisor/hypervisor/arch/x86/notify.c:50 | smp_call_function()
acrn-hypervisor/hypervisor/arch/x86/lapic.c:246 | send_dest_ipi_mask()

**Evidence**
Currently this issue either has no node associated with it or has no details listed.

**Business Risk**
Buffer overflows can be triggered by inputs that are designed to execute code or alter the way the program operates. This may result in erratic program behavior including memory access errors, incorrect results, a crash, or a breach of system security. 

Consequences of buffer overflow include valid data being overwritten and execution of arbitrary and potentially malicious code. For example, buffer overflows can manipulate a program in several ways:

- Overwriting a local variable that is near the buffer in memory to change the behavior of the program to benefit the attacker
- Overwriting the return address in a stack frame so that execution resumes at the return address specified by the attacker 
- Overwriting a function pointer or exception handler, which is subsequently executed
**Recommendation**

Evaluate all object array indexes to not fall out of bounds and implement proper index bounds checking.

lirui34, 2019-07-12:
[External_System_ID] ACRN-3989

Mingyuan18, 2019-07-24:
The patch was merged and pre-integration test no regression issue found:
ACRN TAG: acrn-2019w29.4-150000p
CL: 30400

closed 2019-07-24 by Mingyuan18