projectacrn/acrn-hypervisor #3443: Build ACRN  up2 failed in clearlinux-30340

tomwei666, 2019-07-21:
GSG for Build ACRN up2 section is out of date, need update.
URL is:  https://projectacrn.github.io/latest/tutorials/using_sbl_on_up2.html.
[log.txt](https://github.com/projectacrn/acrn-hypervisor/files/3414208/log.txt)

lirui34, 2019-07-22:
[External_System_ID] ACRN-4803

wenlingz, 2019-07-22:
We will create one PR to fix it

tomwei666, 2019-07-22:
[log1.txt](https://github.com/projectacrn/acrn-hypervisor/files/3417525/log1.txt)

It indeed comipled the file: sos_boot.img   sos_rootfs.img   up2_laag.img.

The output log as shown in the log1.txt. The content in the log1.txt as below, please check whether is something wrong:

mount: laag_image: failed to setup loop device for /home/tom/work/test/up2_laag.img.
.........
losetup: clearlinux.img: failed to set up loop device: No such file or directory
Couldn't install ClearLinux
......
Failed: Exception('No partitions found on /dev/clearlinux.img')
ister create clearlinux.img failed
Failed to create uos images

tomwei666, 2019-07-22:
Flash the system to flash in UP2 board.
When I login followed the step by gsp, cannot launch the uos.The log shows as below:

root@clr-0d98ebe9aca541648337b24751679288 ~ # ./launch_uos.sh -V 1
cpu1 online=1
Launch clearlinux UOS
no /data/clearlinux/clearlinux.img, exit

Is the launch_uos.sh out of date?

leilu, 2019-07-23:
@tomwei666 , we could not reproduce the issue. Please check the create-up2-images.sh is correct.

tomwei666, 2019-07-23:
Although the log ouput errors, the images seem right. The create-up2-images.sh is correct.

Burning the images to up2, loggin the sos, there is something wrong:

$ wget https://raw.githubusercontent.com/projectacrn/acrn-hypervisor/master/doc/tutorials/launch_uos.sh
$ sudo ./launch_uos.sh -V 1

Is this should fork another issue?

leilu, 2019-07-23:
@tomwei666, from your log1.txt , the up2_laag.img is not generated successfully, it makes uos launch failed.

tomwei666, 2019-07-23:
I get clearlinux in docker environment  as below:

docker pull clearlinux:latest  

The clearlinux version is:
root@ubuntu/home/tom/work/123 # cat /etc/os-release 
NAME="Clear Linux OS"
VERSION=1
ID=clear-linux-os
ID_LIKE=clear-linux-os
VERSION_ID=30340
PRETTY_NAME="Clear Linux OS"
ANSI_COLOR="1;35"
HOME_URL="https://clearlinux.org"
SUPPORT_URL="https://clearlinux.org"
BUG_REPORT_URL="mailto:dev@lists.clearlinux.org"
PRIVACY_POLICY_URL="http://www.intel.com/privacy"

You can try to reproduce the issue in the above environment.

leilu, 2019-07-23:
@tomwei666 , please add "--privileged=true" if you want to build in docker environment.

tomwei666, 2019-07-23:
I already add --privileged to  container like this:

docker run -it  --net=host  --privileged --name=clearlinux-acrn  clearlinux:latest
docker start clearlinux-acrn
docker exec -it clearlinux-acrn /bin/bash

Now I execute the shell command one by one in  create-up2-images.sh to locate the reason of compiled error.

tomwei666, 2019-08-03:
I install clearlinux on virtualbox, the create-up2-images.sh can create the image.  After buning image to up2 board, the sos output the error log as enclosure named sos-boot-failure.txt.
PS: I do not connect  panel to up2 board.
[sos-boot-failure.txt](https://github.com/projectacrn/acrn-hypervisor/files/3463632/sos-boot-failure.txt)

tomwei666, 2019-08-03:
Since there is no erro in make images with create-up2-images.sh, I closed this issue. 
I open new issue.

closed 2019-08-03 by tomwei666