projectacrn/acrn-hypervisor #3444: "Breakpoint Exception Exception :EXPECTED_VECTOR : 0x0006 (Invalid Opcode Exception (#UD))" will occur in parsed log when run seed88

lirui34, 2019-07-22:
**Environment:**
**IOC:** Build: 4. 0. 8
**ABL:** rel.1820-Android-HF1 (CSE: FW 3.1.55.2271)
**Build:**
**HV version:**  acrn-2019w10.3-140000p
And cafe git https://github.com/binbinwu1/acrn-hypervisor/tree/support_RIT
café branch:rit_on_acrn-2019w10.3-140000p_new

**Bug detailed description:**
Run café test for seed 88, system will get crash, get café.out log with "python3 errorDump.py", then parse the café.out ,Will get "Breakpoint Exception Exception :EXPECTED_VECTOR : 0x0006 (Invalid Opcode Exception (#UD))" in parsed log information
->>>
Cafe CheckPoints:
Thread 0 => 0x0002001B - Repeated_Setup:RandomCodeEntry
EXPECTED_VECTOR : 0x0006 (Invalid Opcode Exception (#UD)) // The exception that was expected next or at the specified CS:EIP
ACTUAL_VECTOR : 0x0003 (Breakpoint Exception (#BP)) // The exception that occurred
EXPECTED_EIP : 0x000000001828a8b3 // The EIP of the expected exception

**Reproduce steps:**
#. Build image with café patch, and flash onto board successfully.
#. Copy cafe.bin cafe.cfg cafe_cfg.bin cafe_bios.bin cafe_hb.bin cafe.sh errorDump.py parseConfig.py into SOS in /usr/share/acrn/rit/
#. Can SSH to login SOS
#. Run ./café.sh in serial ttyUSB3 port then SUT will get crashed.
#. In SSH to SOS, sync && sync to make the *.dump filed saved
#. In SSH run "cd /usr/share/acrn/rit/; python3 errorDump.py" to get the café.out log
#. Parse the log

**Current result:**
when run seed88, "Breakpoint Exception Exception :EXPECTED_VECTOR : 0x0006 (Invalid Opcode Exception (#UD))" will occur in parsed log

**Expected result:**
Café seed88 test can pass.

lirui34, 2019-07-22:
[External_System_ID] ACRN-3211

Mingyuan18, 2019-08-09:
Verify pass in below build:
ACRN TAG: acrn-2019w31.5-140000p
SOS: 4.19.59-PKT-190722T171340Z   28100(PV version)

closed 2019-08-09 by Mingyuan18