projectacrn/acrn-hypervisor #3454: [Machine Check]Global Machine Check MSR IA32_MCG_STATUS are all available for Kabylake forwards processers.(Write 0 to the reg, it should no exception, actually got #GP)

lirui34, 2019-07-23:
**Environment**
64 bit

**HW/Board**
KBL NUC

**case requirement:**
MCA expose MC common MSRs
ACRN hypervisor shall expose common Machine Check MSRs to any VM. Global Machine Check MSR IA32_P5_MC_ADDR/IA32_P5_MC_TYPE/IA32_MCG_CAP/IA32_MCG_STATUS are all available for Kabylake forwards processers. 

**Description:**
To test IA32_MCG_STATUS register, I write 0H to the IA32_MCG_STATUS register, #GP Exceptions are generated in the hypervisor.
it's ok in the native.

lirui34, 2019-07-23:
[External_System_ID] ACRN-4343

Mingyuan18, 2019-10-10:
acrn-hypervisor:
    repository: https://github.com/projectacrn/acrn-hypervisor.git 
    branch: master index: b16531257ef72431aeb27bfb4f97d96da18202cd 
    date: 2019-09-24
test ok!

closed 2019-10-10 by Mingyuan18