projectacrn/acrn-hypervisor #3467: Potential risk in virtioi_i2c.c & virtio_console.c

conghuic23, 2019-07-25:
Array 'native_path' of size 20 may use index value(s) 20
/home/jenkins_worker/workspace/acrn-daily-kw-scan/acrn-hypervisor/devicemodel/hw/pci/virtio/virtio_i2c.c:446 | native_adapter_create()

Pointer 'portpath' checked for NULL at line 841 may be passed to function and may be dereferenced there by passing argument 1 to function 'virtio_console_open_backend' at line 862.
/home/jenkins_worker/workspace/acrn-daily-kw-scan/acrn-hypervisor/devicemodel/hw/pci/virtio/virtio_console.c:862 | virtio_console_add_backend()

Mingyuan18, 2019-07-25:
[External_System_ID] ACRN-4820

Mingyuan18, 2019-08-08:
The patch was merged and no regression issue found in daily test:
ACRN TAG: acrn-2019w30.5-140000p
CL: 30500

closed 2019-08-08 by Mingyuan18