projectacrn/acrn-hypervisor #3469: [APL NUC] Display goes black while booting; when only one display monitor is connected.

kbillore28, 2019-07-25:
SOS: Clear Linux
UOS: Clear Linux

In APL Nuc, Display goes black while booting; when only one display monitor is connected. It doesn't mean that SOS is NOT booting. you can still connect to SOS through ssh if IP is preknown. Its just that once DRM gets control display goes black.

Issue scenario:
SOS kernel parameter
i915.avail_planes_per_pipe=0x000F00
i915.domain_plane_owners=0x000000001111 
UOS kernel parameter:
i915.avail_planes_per_pipe=0x00000F 

Issue not happens if 
SOS kernel parameter:
i915.avail_planes_per_pipe=0x01010F " 
i915.domain_plane_owners=0x011111110000 " 
UOS kernel parameter:
i915.avail_planes_per_pipe=0x070F00

That means issue happens only when Pipe A is assigned to UOS and NOT to SOS. Also, I Found crash in SOS kernel logs as mentioned below. 

[    1.273161] ------------[ cut here ]------------
[    1.273165] plane 1A assertion failure (expected on, current off)
[    1.273182] SBuf: ele_num=0x1fffe, ele_size=0x20 allocated!
[    1.273209] WARNING: CPU: 3 PID: 143 at drivers/gpu/drm/i915/intel_display.c:1296 assert_plane+0xbf/0xd0
[    1.273211] Modules linked in:
[    1.273217] CPU: 3 PID: 143 Comm: kworker/u8:4 Tainted: G     U            4.19.57-71.iot-lts2018-sos #1
[    1.273219] Hardware name: Intel Corporation NUC6CAYH/NUC6CAYB, BIOS AYAPLCEL.86A.0043.2017.1123.1559 11/23/2017
[    1.273225] Workqueue: events_unbound async_run_entry_fn
[    1.273229] RIP: 0010:assert_plane+0xbf/0xd0
[    1.273231] Code: 84 c0 48 8b 73 18 48 c7 c2 8a 52 68 8c 49 89 c8 48 c7 c7 e0 1d 69 8c 4c 0f 44 c2 45 84 e4 48 0f 45 d1 4c 89 c1 e8 3d 3c a0 ff <0f> 0b e9 74 ff ff ff e8 f5 38 a0 ff 0f 1f 44 00 00 0f 1f 44 00 00
[    1.273232] RSP: 0000:ffff9897f22038a8 EFLAGS: 00010286
[    1.273234] RAX: 0000000000000000 RBX: ffff9897f5966c00 RCX: 0000000000000000
[    1.273236] RDX: 0000000000000000 RSI: ffffffff8b0fffbf RDI: 00000000ffffffff
[    1.273237] RBP: ffff9897f22038c8 R08: 000000004be2f753 R09: ffffffff8cc39cb0
[    1.273238] R10: 0000000000000273 R11: 0000000000033814 R12: 0000000000000001
[    1.273239] R13: ffff9897f4630000 R14: ffff9897f3131000 R15: ffff9897f312b000
[    1.273241] FS:  0000000000000000(0000) GS:ffff9897f7b80000(0000) knlGS:0000000000000000
[    1.273243] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    1.273244] CR2: 0000000000000000 CR3: 0000000240814000 CR4: 00000000003406e0
[    1.273245] Call Trace:
[    1.273256]  intel_atomic_commit_tail+0xa10/0xba0
[    1.273260]  intel_atomic_commit+0x2c5/0x2f0
[    1.273266]  drm_atomic_commit+0x4a/0x50
[    1.273269]  restore_fbdev_mode_atomic+0x1dc/0x1f0
[    1.273273]  restore_fbdev_mode+0x4a/0x160
[    1.273275]  drm_fb_helper_restore_fbdev_mode_unlocked+0x54/0xb0
[    1.273277]  drm_fb_helper_set_par+0x30/0x60
[    1.273281]  intel_fbdev_set_par+0x1a/0x50
[    1.273285]  fbcon_init+0x486/0x600
[    1.273289]  visual_init+0xdc/0x140
[    1.273291]  do_bind_con_driver+0x1bf/0x2b0
[    1.273294]  do_take_over_console+0x12f/0x1a0
[    1.273299]  do_fbcon_takeover+0x5c/0xb0
[    1.273301]  fbcon_event_notify+0x7c0/0x900
[    1.273303]  ? fb_add_videomode+0x84/0x110
[    1.273307]  notifier_call_chain+0x51/0xc0
[    1.273310]  blocking_notifier_call_chain+0x50/0x70
[    1.273312]  fb_notifier_call_chain+0x1b/0x20
[    1.273314]  register_framebuffer+0x215/0x330
[    1.273318]  __drm_fb_helper_initial_config_and_unlock+0x261/0x420
[    1.273320]  drm_fb_helper_initial_config+0x36/0x40
[    1.273323]  intel_fbdev_initial_config+0x18/0x30
[    1.273325]  async_run_entry_fn+0x3e/0x170
[    1.273328]  process_one_work+0x19e/0x3b0
[    1.273331]  worker_thread+0x50/0x3c0
[    1.273333]  kthread+0x12d/0x150
[    1.273335]  ? process_one_work+0x3b0/0x3b0
[    1.273337]  ? kthread_create_worker_on_cpu+0x70/0x70
[    1.273341]  ret_from_fork+0x3a/0x50
[    1.273347] ---[ end trace dd13afa35ad5f63a ]---

lirui34, 2019-07-26:
[External_System_ID] ACRN-4879

minhe1, 2019-07-29:
It's expected behavior since you assigned pipe A to UOS, in  your setting "SOS kernel parameter
i915.avail_planes_per_pipe=0x000F00 i915.domain_plane_owners=0x000000001111 ". When there's only one monitor connected, it will be assigned to pipe A.

Mingyuan18, 2019-08-07:
@kbillore28 , Please close this issue due to it as work as design.

kbillore28, 2019-08-08:
thanks for support. closing

closed 2019-08-08 by kbillore28