projectacrn/acrn-hypervisor #3470: WaaG: Can't boot LTSC-Virtio-Gfx ISO disk in install_vga.sh

ttzeng, 2019-07-25:
I've been following [this tutorial](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html) several times to install the inserted WIn10 LTSC-Virtio-Gfx ISO file in the Boot Manager, but I still can't get it work -- the Boot Manager continuously shows ``ConvertPages: Incompatible memory types, the pages to allocate have been allocated`` message, and refuses to boot the ISO disk image. Any idea what could be the issue?

lirui34, 2019-07-26:
[External_System_ID] ACRN-4880

chejianj, 2019-07-26:
can you please tell which HW platform are you trying?
can you please provide more information about your steps?
can you please paste the OVMF logs here?

ttzeng, 2019-07-29:
@chejianj thanks for looking into this. The followings are more inputs for your reference:
> can you please tell which HW platform are you trying?

NUC7i5DNH

> can you please provide more information about your steps?

I've followed exactly the same instructions in [this tutorial](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#pre-install-drivers-and-re-generate-windows-iso), until the step 6 in section [Install Windows 10](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#install-windows-10).

> can you please paste the OVMF logs here?

Please reference the attached log file.
[install.log](https://github.com/projectacrn/acrn-hypervisor/files/3440316/install.log)

chejianj, 2019-07-29:
Thanks for your feedback.

Please select the device that contains the LTSC-Virtio-Gfx ISO by ENTER **followed by any key**.

Actually you can see "**Press any key to boot from CD or DVD**" after you have selected the device that contains the LTSC-Virtio-Gfx ISO

ttzeng, 2019-07-29:
@chejianj yes, I saw "**Press any key to boot from CD or DVD...**" in the VNC session, but it doesn't boot the ISO file, and returns to the Boot Manager. The console shows "``ConvertPages: Incompatible memory types, the pages to allocate have been allocated``"

chejianj, 2019-07-29:
It shows "ConvertPages: Incompatible memory types, the pages to allocate have been allocated"
You can ignore that.

On my side, it show "ConvertPages: Incompatible memory types, the pages to allocate have been allocated" as well and I press space, the installation can proceed.

ttzeng, 2019-07-29:
Thanks @chejianj , I can successfully install to the ``win10-ltsc-virtio.img`` now. However, after installation, the ``launch_igx-waag.sh`` launch script in the step 1 of [Boot Windows with GVT-g on ACRN](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#boot-windows-with-gvt-g-on-acrn) is missing. I tried to use the [``launch_win.sh``](https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/launch_win.sh) to launch the VM, it can't boot to the Win10 desktop, the display stops after showing several ``modify_bar_registration: bypass for pci-gvt`` messages. Any options required to be passed while launching the WaaG VM?

chejianj, 2019-07-30:
If you have installed the win10 to the image successfully, you can use [launch_win.sh](https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/launch_win.sh)  to launch WaaG.

Before that please make sure you have followed below sections to update all the binaries:
[Build ACRN EFI Images](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#build-acrn-efi-images)
[Build Service OS Kernel](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#build-service-os-kernel)
[Update Kernel on KBL NUC](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#update-kernel-on-kbl-nuc)

Follow below section to config display:
Boot Windows with GVT-g on [ACRN](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#boot-windows-with-gvt-g-on-acrn)

It will take a while before you can see windows desktop on one monitor.

ttzeng, 2019-07-30:
Thanks @chejianj , I was using the latest CL kernel and thought it already merged WaaG patches, it seems not the case. By using the home made kernel, I can run the WaaG now. Thanks for the support!

closed 2019-07-30 by ttzeng