projectacrn/acrn-hypervisor #3477: Add board specific cpu state table to support Px Cx

lirui34, 2019-07-29:
Currently the Px Cx supported SoCs which listed in cpu_state_tbl.c is limited,
    and it is not a wise option to build a huge state table data base to support
    Px/Cx for other SoCs. This patch give a alternative solution that build a board
    specific cpu state table in board.c which could be auto-generated by offline
    tool, then the CPU Px/Cx of customer board could be enabled;
    
    Hypervisor will search the cpu state table in cpu_state_tbl[] first, if not
    found then go check board_cpu_state_tbl. If no matched cpu state table is found
    then Px/Cx will not be supported;

lirui34, 2019-07-29:
[External_System_ID] ACRN-4885

NanlinXie, 2022-05-04:
Closed with no regression issues.

closed 2022-05-04 by NanlinXie