projectacrn/acrn-hypervisor #3488: ACRN on  Intel Nuc 8

limbo127, 2019-07-29:
Hello, 
As we cannot bought < intel nuc8 at this time, 
we are trying to install ACRN on intel nuc8 ( NUC8 BEB i5  latest bios firmawre : 0073 ) without success.
1 - clear os is working well, then install acrn
2 - Nuc is stopping during EfI, I think ( screen display Intel Boot + number of precessor at top 

How can we do ?
Regards, 
Nicolas

tsunghanlin, 2019-07-29:
Same here. Using Intel® NUC Kit NUC7i7DNHE with latest BIOS 0066(DNi70066.bio).

* tried install ACRN following getting start doc.
* tried build ACRN myself

but all got stuck at BIOS with this
"detected 8 processes, 8enabled
current on process 0"
screen.

lirui34, 2019-07-30:
[External_System_ID] ACRN-4895

coneypo, 2019-07-30:
Hi,

NUC8 BEB i5 with i5-8259U is **Coffee lake**, but currently we boot ACRN on **Apolly or Kaby lake**;

For NUC supported, please refer to https://projectacrn.github.io/latest/hardware.html;

coneypo, 2019-07-30:
> Same here. Using Intel® NUC Kit NUC7i7DNHE with latest BIOS 0066(DNi70066.bio).
> 
> * tried install ACRN following getting start doc.
> * tried build ACRN myself
> 
> but all got stuck at BIOS with this
> "detected 8 processes, 8enabled
> current on process 0"
> screen.

Please refer to following BIOS setting (enable VT-x and disable hyper-threading), and boot again:

Item | Setting
-- | --
VMX | Enabled
VT-D | Enabled
Hyper-Threading | Disabled
Secure Boot | Disabled

limbo127, 2019-07-30:
Hello,
Item Setting VMX Enabled VT-D Enabled Hyper-Threading Disabled Secure
Boot Disabled
same result
detect 4 processes,4 enabled
currnt on process 0
then block

coneypo, 2019-07-31:
> Hello, Item Setting VMX Enabled VT-D Enabled Hyper-Threading Disabled Secure Boot Disabled same result detect 4 processes,4 enabled currnt on process 0 then block Le mar. 30 juil. 2019 à 08:38, Xie Zhengtian <notifications@github.com> a écrit :
> […](#)

For NUC7i7DNHE with Kaby Lake i7-8650U, we can mofidy BIOS setting to boot ACRN;
For NUC8i5BEB with Coffee lake i5-8259U you are using, we are not able to support ACRN currently.

We recommend using boards with Kaby Lake or Apolly Lake to boot ACRN and please refer to supported hardware from here: https://projectacrn.github.io/latest/hardware.html;

tsunghanlin, 2019-07-31:
> > Same here. Using Intel® NUC Kit NUC7i7DNHE with latest BIOS 0066(DNi70066.bio).
> > 
> > * tried install ACRN following getting start doc.
> > * tried build ACRN myself
> > 
> > but all got stuck at BIOS with this
> > "detected 8 processes, 8enabled
> > current on process 0"
> > screen.
> 
> We can boot ACRN on Intel NUC Kit **NUC7i7DNHE** with **Kaby Lake i7-8650U**;
> 
> Please refer to following BIOS setting (enable hardware virtualization and disable hyper-threading), and boot again:
> 
> Item	Setting
> VMX	Enabled
> VT-D	Enabled
> Hyper-Threading	Disabled
> Secure Boot	Disabled

thanks for the reply.
checked my BIOS config and found that HT was on and turned it off.
but still no luck.

my config/SKU here:

- Intel NUC Kit NUC7i7DNHE Reference with Intel® Core™ i7-8650U
- BIOS: 0066
- have 1 sata ssd and 1 M.2 ssd.
- Clear OS on M.2(recognized as sub)
- tried install using https://raw.githubusercontent.com/projectacrn/acrn-hypervisor/master/doc/getting-started/acrn_quick_setup.sh
- cat /etc/os-release: 30550
- install hypervisor with command: 「sudo sh acrn_quick_setup.sh -s 30550 -e /dev/sdb1」

still got stuck at 
"detected 4 processes, 4 enabled
current on process 0"

ttzeng, 2019-07-31:
@tsunghanlin how many memory installed in your NUC system?

tsunghanlin, 2019-07-31:
32 GB(DDR4-2400) of them

limbo127, 2019-07-31:
Ok thanks,
Have you any idea about NUC8 support ?
In Europe, Intel NUc < 8 cannot be buy , only NUC8 is now retail.
In second point, we want to base our product on acrn in place of kvm, but
we need to known if acrn will be near of the Intel roadmap processor
Is it a big work to deal with new generation, how can we help ?
Regards,
Nicolas

coneypo, 2019-07-31:
> * Intel NUC Kit NUC7i7DNHE Reference with Intel® Core™ i7-8650U
> * BIOS: 0066
> * have 1 sata ssd and 1 M.2 ssd.
> * Clear OS on M.2(recognized as sub)
> * tried install using https://raw.githubusercontent.com/projectacrn/acrn-hypervisor/master/doc/getting-started/acrn_quick_setup.sh
>   
>   * also tried to modified 2018 string in the script to 2019, but no luck as well
> * cat /etc/os-release: 30550
> * install hypervisor with command: 「sudo sh acrn_quick_setup.sh -s 30550 -e /dev/sdb1」

I found that:
- 1 sata ssd and 1 M.2 ssd.
- Clear OS on M.2(recognized as sub)
- install hypervisor with command: 「sudo sh acrn_quick_setup.sh -s 30550 -e /dev/sdb1」

You have installed Clear Linux on M.2 disk (will be **/dev/nvme0n1xx**), but installed ACRN on SATA disk **/dev/sdbx**;
Can you show your disk info on this NUC with "lsblk"?
Install Clear Linux and ACRN on the same disk and try again;

tsunghanlin, 2019-07-31:
> You have installed Clear Linux on M.2 disk (will be /dev/nvme0n1xx), but installed ACRN on SATA disk /dev/sdbx;
> Can you show your disk info on this NUC with "lsblk"?
> Install Clear Linux and ACRN on the same disk and try again;
sorry for the insufficient info.
the M.2 ssd I used is a SATA one, so it is recognized as /dev/sdxx

anyway, I just removed my SATA ssd and left M.2 SATA ssd along and here is my disk info I retrieved in Clear Linux:
$ cat lsblk.log
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 465.8G  0 disk
├─sda1   8:1    0   142M  0 part
├─sda2   8:2    0   244M  0 part [SWAP]
└─sda3   8:3    0 465.4G  0 part /
$ cat mount.log
/dev/sda3 on / type ext4 (rw,relatime,stripe=256)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)

but still can't boot into ACRN.(tried both kernel-iot-lts2018 and kernel-iot-lts2019)

ttzeng, 2019-07-31:
> 32 GB(DDR4-2400) of them

@tsunghanlin I remember we need to modify the default configuration to support 32GB memory at some point, not sure it's more flexible in recent releases. The NUC7ixDNHE model is supposed OK to run, you may give it a try. :)

ttzeng, 2019-07-31:
@tsunghanlin The prebuilt ``acrn.efi`` is built for [16GB memory](https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/Kconfig#L221), you'd need to rebuild the binary for your platform. The easiest way may be removing one memory bank from your system for verification. :)

janruh, 2019-08-02:
I am running into the same problem with a Xeon W-2133. HT disabled, VT-x, VT-d enabled. I also set the PLATFORM_RAM_SIZE and SOS_RAM_SIZE to 32GB, but ACRN still hangs in the hv_jump function.

coneypo, 2019-08-05:
> Xeon W-2133

**Xeon W-2133** is Sky Lake, and currently ACRN supports Apolly Lake or Kaby Lake;
Please refer to https://projectacrn.github.io/latest/reference/hardware.html for platforms supported.

tsunghanlin, 2019-08-05:
> @tsunghanlin The prebuilt `acrn.efi` is built for [16GB memory](https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/Kconfig#L221), you'd need to rebuild the binary for your platform. The easiest way may be removing one memory bank from your system for verification. :)

hi @ttzeng thx for the suggestion.
however, I have hard time configure and do a custom build for my machine.
could you guys kindly guide my to some docs? or show me how to do a custom build? thx

* I refer to https://projectacrn.github.io/latest/develop/building-from-source.html#build-the-hypervisor-device-model-and-tools
* in top dir, I use 「make BOARD=kbl-nuc-i7」but no acrn.efi generated in build/ dir.

still figuring out this build system though...

anyways,
thanks in advance

ttzeng, 2019-08-06:
> could you guys kindly guide my to some docs? or show me how to do a custom build? thx

@tsunghanlin , It seems you are almost there! If you already edited the default config for your platform, you should ``make FIRMWARE=uefi BOARD=kbl-nuc-i7`` to generate the ACRN hypervisor in UEFI executable ``acrn.efi`` format.

PS: The current build guide does have instructions on [customization for derived boards](https://projectacrn.github.io/latest/getting-started/building-from-source.html#save-as-default-configuration), it just does not call out the use of ``FIRMWARE`` build variable. For ``kbl-nuc-i7`` board, the [Makefile](https://github.com/projectacrn/acrn-hypervisor/blob/master/Makefile#L7-L12) will generate ACRN hypervisor for SBL firmware if that variable is omitted. I'd propose to change that. #3540

tsunghanlin, 2019-08-13:
> > could you guys kindly guide my to some docs? or show me how to do a custom build? thx
> 
> @tsunghanlin , It seems you are almost there! If you already edited the default config for your platform, you should `make FIRMWARE=uefi BOARD=kbl-nuc-i7` to generate the ACRN hypervisor in UEFI executable `acrn.efi` format.
> 
> PS: The current build guide does have instructions on [customization for derived boards](https://projectacrn.github.io/latest/getting-started/building-from-source.html#save-as-default-configuration), it just does not call out the use of `FIRMWARE` build variable. For `kbl-nuc-i7` board, the [Makefile](https://github.com/projectacrn/acrn-hypervisor/blob/master/Makefile#L7-L12) will generate ACRN hypervisor for SBL firmware if that variable is omitted. I'd propose to change that to avoid confusion. #3541

Thanks for the tip. Somehow I just cannot use my own built acrn.efi to boot the system.
However, after removing 16GB of my memory and referring to the latest document to automatically install the hypervisor, I can now enter the SOS. (Guess now I have to dig more about the whole arch to figure out what was going on about my own built...)

Anyway, thanks again.

paspndevops, 2019-10-18:
Hi,

I am trying to install ACRN on intel nuc ( NUC7i7DNK(BOARD-NUC7i7DNB) BIOS FW-DNKBLi7v.86A.0056.2018.1003.1622 ) without success.

1 - clearlinux is installed and working well, then installed acrn (27960)

2 - Nuc is stopping during EfI, screen display Intel Boot + number of precessor at top

**"detected 4 processes, 4 enabled
current on process 0"**

khyberspache, 2019-11-25:
I've spent a while trying to get this running on a **Kaby Lake** NUC8i7HNB (it's the first generation of NUC8s, so it *should* work), but I have not been able to Get any configuration to boot past the below.  I attached the board config as a file.
```
detected 4 processes, 4 enabled
current on process 0
```
I have tried:
1. Default quick install script with a NUC7 board config (it failed as I assumed it would).
2. Custom built several EFIs...
- Building the acrn hypervisor with a generic board config w/ 32 GB RAM configuration
- Built a board specific configuration using the `board_config` tool and the SDC scenario (tried several variations of this)

General build approach is to use the quick config script to download the correct service OS version and replace the EFI image with my custom built images and attempt to boot (I'm only trying to get the SDC scenario booted, so this should be supported).  Are there other installation methods to try?

I've follow pretty much every build instruction in the Docs to no avail.  What would you recommend as next steps?  I did notice the board config produced by the script will cause the board_cfg_gen script to fail because of these two lines:
```
	Region 0: [virtual] Memory at 2000000000 (64-bit, prefetchable) [size=4G]
	Region 2: [virtual] Memory at 2100000000 (64-bit, prefetchable) [size=2M]
```
That is why I tried to do the generic board approach.  I tried patching the cfg_script to handle those lines (and tried both removing the lines and modifying them to non-virtual address spaces).

Pretty much have no idea where to go from here... any tips are appreciated. 

[NUC8i7HNB-Full.txt](https://github.com/projectacrn/acrn-hypervisor/files/3888515/NUC8i7HNB-Full.txt)

fuzhongl, 2020-06-28:
RAM_SIZE beyond  16GB  issue is fixed in ACRN 1.5

closed 2020-06-28 by fuzhongl