projectacrn/acrn-hypervisor #3529: sos kernel crash

tomwei666, 2019-08-04:
clearlinux: 29750
1. Getting the create-up2-images.sh file from  https://raw.githubusercontent.com/leilu/acrn-hypervisor/issue_3443/doc/tutorials/create-up2-images.sh.
2. uos.json content is as below:
{
    "DestinationType" : "virtual",
    "PartitionLayout" : [ { "disk" : "clearlinux.img", "partition" : 1, "size" : "100M", "type" : "EFI" },
                          { "disk" : "clearlinux.img", "partition" : 2, "size" : "10G", "type" : "linux" } ],
    "FilesystemTypes" : [ { "disk" : "clearlinux.img", "partition" : 1, "type" : "vfat" },
                          { "disk" : "clearlinux.img", "partition" : 2, "type" : "ext4" } ],
    "PartitionMountPoints" : [ { "disk" : "clearlinux.img", "partition" : 1, "mount" : "/boot" },
                               { "disk" : "clearlinux.img", "partition" : 2, "mount" : "/" } ],
    "Version": 29750,
    "Bundles": ["kernel-iot-lts2018", "openssh-server", "x11-server", "os-core", "os-core-update"]
}
3. ./create-up2-images.sh --images-type all --clearlinux-version 29070 --laag-json uos.json
4. burn the images to up2 board
5. When up2 board boots,the kernel crash.

burn-image.txt:  burn image to up2 board,the console output log
sos-boot-failure.txt:  kernel crash log
[burn-image.txt](https://github.com/projectacrn/acrn-hypervisor/files/3464515/burn-image.txt)
[sos-boot-failure.txt](https://github.com/projectacrn/acrn-hypervisor/files/3464516/sos-boot-failure.txt)

Mingyuan18, 2019-08-07:
[External_System_ID] ACRN-4950

fuzhongl, 2019-09-03:
@tomwei666  Please follow this doc:(Newer Clear Linux image should be used:such as 30690)
https://projectacrn.github.io/latest/tutorials/using_sbl_on_up2.html
Thanks!

tomwei666, 2019-09-04:
Follow the url:https://projectacrn.github.io/latest/tutorials/using_sbl_on_up2.html ，using the 30690 version. Then burn the image to up2 board, sos throw other error:

Starting MB Kernel ...

ACRN Hypervisor
calibrate_tsc, tsc_khz=1094400
[7123496us][cpu=0][sev=2][seq=1]:HV version 1.2-unstable-2019-08-05 10:20:41- DBG (daily tag:) build by mockbuild, start time 7095511us
[7137099us][cpu=0][sev=2][seq=2]:API version 1.0
[7142128us][cpu=0][sev=2][seq=3]:Detect processor: Intel(R) Celeron(R) CPU N3350 @ 1.10GHz
[7151301us][cpu=0][sev=1][seq=4]:SECURITY WARNING!!!!!!
[7157017us][cpu=0][sev=1][seq=5]:Please apply the latest CPU uCode patch!
[7173879us][cpu=0][sev=2][seq=19]:Start VM id: 0 name: ACRN SOS VM
ACRN:\>[7459187us][cpu=0][sev=3][seq=22]:vlapic: Start Secondary VCPU1 for VM[0]...

ACRN:\>
ACRN:\>vm_console

----- Entering VM 0 Shell -----
[    2.403601] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    2.404366] CPU: 0 PID: 1 Comm: swapper/0 Tainted: G     U            4.19.62-73.iot-lts2018-sos #1
[    2.404840] Call Trace:
[    2.404987]  dump_stack+0x8e/0xca
[    2.405171]  ? rest_init+0x27/0xc1
[    2.405359]  panic+0xec/0x285
[    2.405525]  ? do_execve+0x25/0x30
[    2.405711]  ? putname+0x4a/0x50
[    2.405889]  ? rest_init+0xc1/0xc1
[    2.406075]  kernel_init+0xfb/0x106
[    2.406266]  ret_from_fork+0x3a/0x50
[    2.406909] reboot: panic mode set: p,w
[    2.407140] Kernel Offset: 0x6000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[    2.410812] Rebooting in 10 seconds..

Thanks!

leilu, 2019-09-10:
@tomwei666 , could you please provide your create-up2-images.sh, uos.json, and the output when run the `bash -x create-up2-images.sh --images-type all --clearlinux-version 30690 --laag-json uos.json `

tomwei666, 2019-10-14:
I redo the above steps, This bug does not appear!
This bug can clsoe.
Thanks!

closed 2019-10-14 by tomwei666