projectacrn/acrn-hypervisor #3609: Host entered standby state with power key flickering in coldboot test.

Mingyuan18, 2019-08-22:
Environment:
ClearLinux version: 30300
ACRN tag: acrn-2019w28.2-140000p
HW: KBLNUC DNHi7
Bug detailed description:
Host entered standby state with power key flickering in coldboot test, failed rate: 1/200
Reproduce steps:
1. Before begin coldboot test,enable no passwd ssh for SOS and RTVM.
2. Boot up SOS
3. Boot up Zephyr guest with UUID "d2795438-25d6-11e8-864e-cb7a18b34643"
4. Boot up RTVM with UUID "495ae2e5-2603-4d64-af76-d4bc5a8ec0e5"
5. Reboot RTVM and then will restart the whole system
6. After SOS boot up, return to step 3
Current result:
Host entered standby state with power key flickering in coldboot test, failed rate: 1/200
Expected result:
Host can reboot successfully every time in the coldboot test.

Mingyuan18, 2019-08-22:
[External_System_ID] ACRN-4631

fuzhongl, 2019-08-28:
Following patch is under review; will PR late.
"
diff --git a/hypervisor/arch/x86/pm.c b/hypervisor/arch/x86/pm.c
index f7320314..a8cdbd62 100644
--- a/hypervisor/arch/x86/pm.c
+++ b/hypervisor/arch/x86/pm.c
@@ -208,6 +208,11 @@ void reset_host(void)
 	/* TODO: gracefully shut down all guests before doing host reset. */
 
 	/*
+	 * Assumption:
+	 * The platform we are running must support at least one of reset method:
+	 *   - ACPI reset
+	 *   - 0xcf9 reset
+	 *
 	 * UEFI more likely sets the reset value as 0x6 (not 0xe) for 0xcf9 port.
 	 * This asserts PLTRST# to reset devices on the platform, but not the
 	 * SLP_S3#/4#/5# signals, which power down the systems. This might not be
@@ -217,22 +222,14 @@ void reset_host(void)
 		(gas->bit_width == 8U) && (gas->bit_offset == 0U) &&
 		(gas->address != 0U) && (gas->address != 0xcf9U)) {
 		pio_write8(host_reset_reg.val, (uint16_t)host_reset_reg.reg.address);
+	} else {
+		/* making sure bit 2 (RST_CPU) is '0', when the reset command is issued. */
+		pio_write8(0x2U, 0xcf9U);
+		udelay(50U);
+		pio_write8(0xeU, 0xcf9U);
+		udelay(50U);
 	}
 
-	/*
-	 * Fall back
-	 * making sure bit 2 (RST_CPU) is '0', when the reset command is issued.
-	 */
-	pio_write8(0x2U, 0xcf9U);
-	pio_write8(0xeU, 0xcf9U);
-
-	/*
-	 * Fall back
-	 * keyboard controller might cause the INIT# being asserted,
-	 * and not power cycle the system.
-	 */
-	pio_write8(0xfeU, 0x64U);
-
 	pr_fatal("%s(): can't reset host.", __func__);
 	while (1) {
 		asm_pause();
"

closed 2019-10-10 by Mingyuan18