projectacrn/acrn-hypervisor #3610: LaaG hang while SOS run daimleric and LaaG run workloads loop with zephyr idle.

Mingyuan18, 2019-08-22:
Environment:
ClearLinux version: 30580
HW: KBLNUC DNHi7
Bug detailed description:
LaaG hang while SOS run daimleric and LaaG run workloads loop with zephyr idle.
LaaG printed the log info "[drm:gen8_de_irq_handler] ERROR The master control interrupt lied (DE PIPE)!" continually.
Reproduce steps:
1. Enable no passwd ssh for SOS and UOS, and enable UOS auto boot.
2. Connect with $SOS_IP & $UOS_IP by ssh.
3. In SOS, continually run daimeric.
4. In UOS, continually and repeatedly run workloads loop, including glmark,gfxbench,video_playback,stress-ng,stress,iozone,iperf,fio,stress_wayland,MSDK,Geekbench.
Current result:
LaaG hang while SOS run daimleric and LaaG run workloads loop with zephyr idle.
LaaG printed the log info "[drm:gen8_de_irq_handler] ERROR The master control interrupt lied (DE PIPE)!" continually.

Mingyuan18, 2019-08-22:
[External_System_ID] ACRN-4984

fuzhongl, 2019-08-23:
Need revert following patch from acrn-kernel:
revert bbb891728d82834ec450f6a61792f715f4ec3013

closed 2019-10-10 by Mingyuan18