projectacrn/acrn-hypervisor #3630: Clean up the code on drm/i915/gvt.

Mingyuan18, 2019-08-30:
1. A bit in GFX_MODE register could disable HW privilege check on commands from non-privilege batch buffers. Malicious guest can set this bit to allow to run privileged commands in non-privilege batch buffer.
2. Bits in CSFE_CHICKEN1 register could allow RCS/VCS/BCS access to other engines’ non-privilege registers. Malicious guest can enable the access to get information of other context or break other context workload.
3. MI_FLUSH_DW, PIPE_CONTROL’s index mode writes host engine status page instead of per-context status page, guest may corrupt global status page.
4. intel_gvt_ggtt_validate_range() only check memory’s start and end pointer in aperture OR in hidden range, a large buffer set to cover across this two ranges may be set.

Mingyuan18, 2019-08-30:
[External_System_ID] ACRN-5108

fuzhongl, 2019-11-19:
@liujunming Could you please close this git issue?
Thanks!

Mingyuan18, 2019-11-19:
The patch was merged and no regression issue found.

closed 2019-11-19 by Mingyuan18