projectacrn/acrn-hypervisor #3654: NUC7i7BNH not supported anymore

ngaloppo, 2019-09-06:
Despite what is listed [in the latest ACRN documentation](https://projectacrn.github.io/latest/reference/hardware.html#verified-platforms-according-to-acrn-usage), KabyLake NUC7i7BNH doesn't seem to be supported anymore. Looking at the hypervisor code in `arch/x86/configs` it looks like it has been superseded by KabyLake NUC7i7DNH. 

Is the ACRN team still validating NUC7i7BNH? I have such a NUC with 32GB memory, but it doesn't boot. I tried the following: 

1. Remove one stick of RAM, effectively booting with 16GB. Boot process halts at the BIOS screen, doesn't get into UEFI boot menu.
2. Custom build `acrn.efi` by adjusting the target RAM size to 32GB as described [here](https://projectacrn.github.io/latest/faq.html#how-do-i-configure-acrn-s-memory-use). Here I do get the UEFI boot menu, but when booting into ACRN, the system automatically resets after the following dmesg console output: `"fb: switching to inteldrfmfb...`

Mingyuan18, 2019-09-09:
[External_System_ID] ACRN-5141

fuzhongl, 2019-09-09:
@ngaloppo  Which scenario are you using? We will try to reproduce it on KabyLake NUC7i7BNH.
Thanks!

ngaloppo, 2019-09-09:
@fuzhongl using the SDC default scenario

ngaloppo, 2019-09-17:
@fuzhongl Any progress on this topic?

fuzhongl, 2019-09-18:
> @fuzhongl Any progress on this topic?

@ngaloppo 
Which CL build are you used?
Try with CL 30970 on KabyLake NUC7i7BNH with 8GB memory; can't reproduce this issue. Will try 8+8 GB when it is available. 
Thanks!

fuzhongl, 2020-05-05:
@ngaloppo The issue about memory greater than 16G has been fixed since v1.5, please have a try.
Thanks!

closed 2020-06-28 by fuzhongl