projectacrn/acrn-hypervisor #3673: Incorrect reference to OVMF.fd in sample UOS startup script

ngaloppo, 2019-09-11:
The sample UOS startup script references `OVMF.fd` in the current directory ([here](https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/launch_uos.sh#L110)) but this is incorrect (in Clear Linux SOS it is located in `/usr/share/acrn/bios/OVMF.fd`.

gvancuts, 2019-09-12:
Thanks @ngaloppo ! I was considering whether the proper course of action would be to fix the script or add instructions to the Gettting Started Guide to copy the bios file to the local folder where the script it (this is already what we do for Zephyr and VxWorks for example). But unless you feel otherwise, I will make the script point at `/usr/share/acrn/bios/OVMF.fd`.

@lirui34 , why do we have another `OVMF.fd` bios file under `doc/tutorials`? Could we use the one we already have (under `devicemodel/bios`) and which gets automatically installed on the Service VM filesystem to boot WaaG? Thanks

Mingyuan18, 2019-09-16:
[External_System_ID] ACRN-5174

lirui34, 2019-09-16:
> @lirui34 , why do we have another `OVMF.fd` bios file under `doc/tutorials`? Could we use the one we already have (under `devicemodel/bios`) and which gets automatically installed on the Service VM filesystem to boot WaaG? Thanks

This `OVMF.fd` bios file under `doc/tutorials` is no more needed since it's functional with the one under `devicemodel/bios` as well.

ngaloppo, 2019-09-17:
@NanlinXie was anyone assigned to the documentation ask?

fuzhongl, 2019-09-17:
> Thanks @gvancuts. Can we also add some documentation on how this all works?
> 
> * For example, how do we configure the boot options line with OVFM? Previously it was done as an option to `acrn-dm`.
> * Does the use of OVFM require a bootloader / EFI configure step?

@ngaloppo 
About launch UOS with OVMF; the location of cmdline is UOS specific: We have "User VM Tutorials"; all information is there: you can also check the comments in launch script.
 a) the cmdline is into loader/entries/xxx.conf of EFI partation for LaaG;
 b) the cmdline is into  /etc/grub.cfg for Ubuntu/detian UOS
 c) OVMF will launch bootloader with /EFI/boot/bootx86.efi

If it isn't clear enough, please let us know!
Thanks!

ngaloppo, 2019-09-20:
@fuzhongl I think that the missing link (for me) is that OVFM uses the EFI boot partition on the UOS image file. This means that:
1. I need to create a loopback device from the uos image
2. Mount the loopback device
3. Edit the EFI entries manually (maybe use `efibootmgr` with the `-d DISK` option?)
4. On Clear Linux UOS, somehow do the equivalent of `clr-boot-manager`? 
5. ...  ?

None of that information is obvious to me in the "User VM Tutorials". I think we need a separate piece of documentation that explains the intent and benefits of the OVMF feature, how to enable it for each of the different flavors of UOS, and how to upgrade a UOS image that is loaded with OVMF.

gvancuts, 2019-09-26:
Hi @ngaloppo , thanks for your feedback. I have discussed this with @NanlinXie too and we agree it would be good to add this to our documentation. I'll file a feature request to track this however as we do not have anyone assigned to it at the moment.

(I will see if I can free up some of my time over the next week or so to do this but no promise).

closed 2019-09-26 by ngaloppo