projectacrn/acrn-hypervisor #3718: WaaG hang after keep WaaG idle in RTVM Create/Destroy test.

dandanyx, 2019-09-23:
Environment:
------------
HW: KBLNUC DNHi7

Reproduce steps:
-------------
1. Before begin coldboot test,enable no passwd ssh for SOS and RTVM.
2. Boot up SOS
3. Boot up WaaG
4. Boot up RTVM
5. Confirm that WaaG/RTVM all boot up by check "ping WaaGIP and ping RTVMIP", and then shutdown RTVM by ssh connected with RTVMIP
6. After RTVM shutdown, then boot up RTVM again, go to step 4.

Current result:
--------------
WaaG hang after keep WaaG idle for 0.5h in RTVM Create/Destroy test.
WaaG displayed normally but mouse and keyboard cannot work. WaaG status displayed "started" by execute "acrnctl list" in SOS or "vm_list" in HV. 

Expected result:
--------------
WaaG can keep alive and work normally while keep WaaG idle in RTVM Create/Destroy test.

Mingyuan18, 2019-09-24:
[External_System_ID] ACRN-5113

Mingyuan18, 2019-10-23:
Can't reproduce this issue in below build:
CL: 31290
ACRN TAG: acrn-2019w41.4-140000p

closed 2019-10-23 by Mingyuan18